aritchie/bluetoothle #284: Problem with partial BleAdapterConfiguration for iOS adapter init

goiditdev, 2019-01-28:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
noted with iOS 12, probably others. Does not seem iOS version specific

## Version of Library
seen in 6.2.3, not present in 6.0.3

## Expected Behaviour
When initializing an adapter on iOS, the two options are to provide a config or not provide a config in the call to CrossBleAdapter.Init. In version 6.0.3 there used to be default values assigned to each field in the BleAdapterConfiguration, but they were removed, as was the default config. In our code we used to create a config object (which gave it the defaults) and then only changed the values we wanted differently, for example:

BleAdapterConfiguration adapterconfig = new BleAdapterConfiguration();
adapterconfig.ShowPowerAlert = true;
CrossBleAdapter.Init(adapterconfig);

But when using 6.2.3 this seems to no longer work as expected and instead the adapter status always reports AdapterStatus.Unsupported.

The expected behavior is that the AdapterContext c'tor for Apple, which consumes the config, sets appropriate values, if these aren't provided. It does seem to do so, based on our code review, but something is still amiss. When we change our code to 

BleAdapterConfiguration adapterconfig = new BleAdapterConfiguration();
adapterconfig.ShowPowerAlert = true;
adapterconfig.DispatchQueue = new CoreFoundation.DispatchQueue("acrble");
adapterconfig.RestoreIdentifier = "acr";
CrossBleAdapter.Init(adapterconfig);

Then the adapter works again. Our apologies if we missed something!

## Actual Behavior
Please see the expected behavior description above. Thank you very much for looking into this! We will try to dig in some more as well.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-01-29:
Do you mean the AdapterStatus is reporting Unknown?  This usually means the adapter is not quite ready to work with.  You can monitor the status for PoweredOn or use ScanExtra which does this for you.

mhempel247, 2019-01-29:
No, Unsupported, not unknown. This isn't a startup issue of the adapter, as it persists indefinitely.

aritchie, 2019-01-30:
Can you please submit a full code sample of this.  I haven't experienced the issue with Unsupported.

aritchie, 2019-01-30:
You won't be able to build the Plugin.BluetoothLE project due to multitargeted builds which isn't supported by VS4mac yet.  You can use jetbrains rider on mac or obviously VS4win.

As for this issue, thanks for the repo.  I don't see anything obvious at the moment.

goiditdev, 2019-01-30:
Thank you for the VS info. I hope MS gets both VS up to par soon. 
I experimented a bit more with the issue and it turns out that it is the RestoreIdentifier. More specifically, the value I set there. Here are some examples:

```
//adapterconfig.RestoreIdentifier = "pluginbluetoothle";//fails when uncommented
//adapterconfig.RestoreIdentifier = "acr";              //works fine when uncommented
//adapterconfig.RestoreIdentifier = "acr2";             //works fine when uncommented
//adapterconfig.RestoreIdentifier = "pluginbluetoothl"; //works fine when uncommented
//adapterconfig.RestoreIdentifier = "pluginbluetoothle2"; //works fine when uncommented, so not the length of the identifier (this is longer)
//adapterconfig.RestoreIdentifier = "pluginbluetoothl2"; //works fine when uncommented, so not the length of the identifier (this is same length, different character at end)
//also fails when none are uncommented, because the default value is "pluginbluetoothle"
```

I have no idea what is responsible for that! I even uninstalled all other apps that may use BLE on the development devices we use. We've also tested it on an iPhone 6 and an iPhone X, and it results in the same error there as well.

liu12010, 2019-03-07:
Same Problem here.
But if run without `CrossBleAdapter.Init()` in the AppDelegate
the AdapterStatus will always be unknown before I show the alert dialog via
`await UserDialogs.Instance.AlertAsync(alertConf);`
once I showed the dialog the AdapterStatus come to PowerOn immidiately,then everything working fine as usual.

aritchie, 2019-03-07:
@liu12010 "Same" replies are unhelpful.  This is a non-critical issue which I will look into with a future version.  In the meantime, you should be using Adapter.WhenStatusChanged() or ScanExtra in any case.

aritchie, 2019-05-22:
This is resolved in Shiny.BluetoothLE

closed 2019-05-22 by aritchie