aritchie/bluetoothle #285: iOS Client (Obj-C) "Unlikely Error"

BolletuH, 2019-01-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
12.1.2

## Version of Library
6.2.3

## Expected Behaviour
Read characteristic to make request from server, and respond with appropriate data.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Type:
Plugin.BluetoothLE.BleException

Exception:
Error Domain=CBATTErrorDomain Code=14 "Unlikely error." UserInfo={NSLocalizedDescription=Unlikely error.}

Stacktrace:
  at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.AnonymousObserver`1[T].OnErrorCore (System.Exception error) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.AutoDetachObserver`1[T].OnErrorCore (System.Exception exception) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <370f6a6bb34048878534065376a195cb>:0 
  at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass12_0.<Read>b__1 (System.Object sender, CoreBluetooth.CBCharacteristicEventArgs args) [0x0003e] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Apple\GattCharacteristic.cs:61 
  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.UpdatedCharacterteristicValue (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBCharacteristic characteristic, Foundation.NSError error) [0x00012] in /Library/Frameworks/Xamarin.iOS.framework/Versions/12.2.1.12/src/Xamarin.iOS/CoreBluetooth/CBPeripheral.g.cs:569 
  at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Library/Frameworks/Xamarin.iOS.framework/Versions/12.2.1.12/src/Xamarin.iOS/UIKit/UIApplication.cs:79 
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x0002c] in /Library/Frameworks/Xamarin.iOS.framework/Versions/12.2.1.12/src/Xamarin.iOS/UIKit/UIApplication.cs:63

I receive this error when performing a Read or Write to my Uwp Server via an iOS based device. However, WriteWithoutResponse appears to be functioning as the subscribed method is executed. Looking through repository I could not see where Write and Read were implemented via iOS platform for the GattCharacteristic class.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

BolletuH, 2019-01-29:
Whoops... you are correct that I was examining the "iOS" portion of the repository instead of Apple. Should have paid more attention to the stack trace.

Here is the call within the ViewModel to perform the data transmission.

```
await bluetoothClient.SendAction(StateService.Device, App.Guids[nameof(Constants.RELAY_GUID)], characteristic =>
            {
                characteristic.Write(Encoding.UTF8.GetBytes(message)).Subscribe(writeResult =>
                {
                     //Never reaches here (error is thrown).
                });
            });
```

Here is the code surrounding the `bluetoothClient`.

```        
public Task SendAction(IDevice device, Guid charactersticGuid, Action<IGattCharacteristic> subscribeAction)
       {
            return Task.Run(() =>
            {
                device?.GetKnownCharacteristics(ServiceGuid, charactersticGuid).Subscribe(subscribeAction);
            });
        }
```

Nothing more than a wrapper with the action being defined in the ViewModel and passed as an argument.

aritchie, 2019-01-29:
So there is a whole whack of things going wrong here
1) Oh, and message is nothing more than a JSON serialized string - default with most BLE connections is 20 bytes in either direction. 

2) While you are coding around the RX style, things like Task.Run are unnecessary and going to cause you grief long term.  

3) Again with the RX style, subscribe does have an OnError that you should hook for your apps sake

Please note that I do not do support stuff like this.  The library is working as expected for all known test cases that I have.  I do offer consultancy if you require.

closed 2019-01-29 by BolletuH