aritchie/bluetoothle #289: Device uuid

bloempot71, 2019-02-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 12.1.4
Android 8.0

## Version of Library
6.2.3

## Expected Behaviour
I want to have one phone connect to another phone.
To do that I show a QR Code on phone A with the bluetooth device uuid and scan that with phone B so they can find each other and connect.

I'd like to know how I can find the uuid on phone A that is listed in the scan result of the Adapter.Scan() on phone B. So something like CrossBleAdapter.Current.DeviceUuid

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-02-15:
There isn’t such a thing.  You can advertise a custom service uuid and scan for that though

bloempot71, 2019-02-15:
Do you mean this?

BluetoothServer = await BluetoohAdapter.CreateGattServer();
Plugin.BluetoothLE.Server.IGattService service = BluetoothServer.CreateService(guidService, true);

And then scan with a ScanConfig?

ScanConfig sc = new ScanConfig() {
	ServiceUuids = new List<Guid>() { searchForServiceUuid },
	ScanType = BleScanType.Background,
	AndroidUseScanBatching = true
};

Because I tried that and it didn't find the other phone.

aritchie, 2019-02-15:
It looks right.  I can't formulate a guess without reproducible case.

Please note, this isn't a support channel.  I do offer consultancy if you need help getting this going.

closed 2019-02-17 by aritchie