aritchie/bluetoothle #291: [Help-Needed] 26 byte advertising payload not being detected

slushpuppy, 2019-02-20:
Hi,

     I got a 26 byte packet advertising payload that isn't picked up during the scanning phase.

Removing 7 bytes off the manufacturing data allowed the beacon to be picked up again. What could be the possible causes?

         CrossBleAdapter.Current.WhenStatusChanged().Subscribe(status => {
                   if (status == AdapterStatus.PoweredOn)
                {
                    CrossBleAdapter.Current.Scan(
                          new ScanConfig 
                          {
                              ScanType = BleScanType.LowLatency
                          }
                        ).Subscribe(scanResult => {
                            App.Helper.Debug.WriteJson(scanResult);
                           

                        });
                }
            });

The device is indeed visible with the 26 byte payload in NRF connect

BolletuH, 2019-02-20:
If you're using Bluetooth LE 4.2- then you only have 20-23 bytes of attribute data with the rest being overhead in the PDU. However, if you're using Bluetooth LE 4.2+ then you have up to 251 bytes of payload with 242 bytes of attribute data for transmission.

This should expand a bit on things: https://stackoverflow.com/questions/38913743/maximum-packet-length-for-bluetooth-le

aritchie, 2019-02-20:
Please note, this is not a support channel.  You can file defects with the proper issue template and reproducible case.  I do offer paid consultancy if needed.

If you take a look at the scan code, you'll notice I don't do anything to hijack the process.  I have noticed that apple tends to nullify the manufacturer data where iBeacon gets involved because they have their own API they want to use.

@BolletuH That MTU size is with regards to the GATT characteristic, not the advertising packet.

BolletuH, 2019-02-20:
> Please note, this is not a support channel. You can file defects with the proper issue template and reproducible case. I do offer paid consultancy if needed.
>
> If you take a look at the scan code, [...]

Yep, realized that the moment I posted haha...

slushpuppy, 2019-02-21:
Thanks so much, should have checked bluetooth spec of the phone >.<

closed 2019-02-21 by slushpuppy