aritchie/bluetoothle #292: Ability to set custom PeripheralConnectionOptions on Connect for iOS?

raindeer, 2019-02-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
All

## Version of Library
-

## Expected Behaviour
Would be nice to be able to supply a custom PeripheralConnectionOptions object to the device.Connect() method

## Actual Behavior
Now all properties of PeripheralConnectionOptions are set to true when connecting which yields undesired behavior in some cases.

NotifyOnDisconnection = true,
NotifyOnConnection = true,
NotifyOnNotification = true

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-02-21:
> Now all properties of PeripheralConnectionOptions are set to true when connecting which yields undesired behavior in some cases.

Such as?

raindeer, 2019-02-21:
For instance, with NotifyOnNotification = true iOS prompts the user with an alert box as soon as the app receives a notification on any characteristic while the app runs in the background. In my case I stream data from the device so notifications are received constantly which means you cannot exit the app without getting an alert. And if you close the alert a new one is shown immediately.

raindeer, 2019-02-21:
https://developer.apple.com/documentation/corebluetooth/cbconnectperipheraloptionnotifyonnotificationkey

MKGNZ, 2019-02-21:
Strange as I'm using this library and never had this issue. 2 things with your comment @raindeer 
First from the documentation "This key is useful for apps that have **not** specified the bluetooth-central background mode and cannot display their own alert. " do you have that key in your info.plist? to use this library you should have it defined.
Second 

> In my case I stream data from the device so notifications are received constantly which means you cannot exit the app without getting an alert. 
> 

if you haven't defined "bluetooth-central" in your info.plist then your app doesn't work on the background but it get's suspended. That means your app is **NOT** streaming anything at that point.
From the documentation that you referenced: "A Boolean value that specifies whether the system should display an alert for all notifications received from a given peripheral **if the app is suspended at the time**." So your app doesn't stream in the background. 

Define "bluetooth-central" background mode and see what happens.

raindeer, 2019-02-22:
Im aware of the bluetooth-central issue (I reported #288 :)) And I dont want to add it to the plist, because I don't need to have a connection to the device when the app is closed. However, if you exit an app briefly the connection is not killed right away. My use case is that I want users so exit the app for a short while and then go back without needing to connect again. This is not possible with this lib because an alert is shown, forcing the user to enter the app again.
This worked well in my previous implementation based directly on Core Bluetooth.
This lib is great because I can share code between Android and iOS but the lack of configuration options limits the use case.

aritchie, 2019-05-22:
This is implemented in Shiny.BluetoothLE - if something limits you in the future, feel free to send a PR ;)

closed 2019-05-22 by aritchie