aritchie/bluetoothle #296: Null Reference Exception when getting characteristic on Android for some peripherals

Carbaholic, 2019-03-04:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Lollipop

## Version of Library

## Expected Behaviour
I would expect an exception that tells me what is null and why

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Null reference exception

Stack Trace: 

  at Plugin.BluetoothLE.GattCharacteristic..ctor (Plugin.BluetoothLE.IGattService service, Plugin.BluetoothLE.Internals.DeviceContext context, Android.Bluetooth.BluetoothGattCharacteristic native) [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\GattCharacteristic.cs:25 
  at Plugin.BluetoothLE.GattService+<>c__DisplayClass4_0.<GetKnownCharacteristics>b__0 (System.IObserver`1[T] ob) [0x0004a] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\GattService.cs:48 
  at System.Reactive.AnonymousObservable`1[T].SubscribeCore (System.IObserver`1[T] observer) [0x00000] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.ObservableBase`1[T].Subscribe (System.IObserver`1[T] observer) [0x00037] in <370f6a6bb34048878534065376a195cb>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.ExceptionHelpers.ThrowIfNotNull (System.Exception exception) [0x0000d] in <370f6a6bb34048878534065376a195cb>:0 
  at System.Reactive.Subjects.AsyncSubject`1[T].GetResult () [0x00039] in <370f6a6bb34048878534065376a195cb>:0 
  at Lynkd.Models.Peripherals.Peripheral+<InitPeripheral>d__110.MoveNext () [0x00147] in /Development/Mobile/lynkd_mobile/lynkd-models/Lynkd.Models/Lynkd.Models/Peripherals/_Peripheral.cs:926 

I am simply connecting to a peripheral and reading a characteristic that works fine on iOS. 

    `         IGattService service = await device.GetKnownService(BleService.DeviceInformation).Timeout(timeout);
                IGattCharacteristic characteristic = await service.GetKnownCharacteristics(BleCharacteristic.DeviceProtocol).Timeout(timeout);

                HardwareTypeResult = await characteristic.Read().Timeout(timeout);`

aritchie, 2019-03-04:
You can use sourcelink to debug this.  I haven’t seen this issue before

Carbaholic, 2019-03-06:
I haven't gotten sourcelink going yet but I have learned that this occurs when you try to communicate with a peripheral that should be paired but is not paired. 

This is my first week working with bluetooth for android so I'm quickly learning the differences. I'm guessing I just need to check the pairing status and request unpaired devices to pair....which didn't work on iOS.