aritchie/bluetoothle #297: Android 6.0 Will not Pair

Carbaholic, 2019-03-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android M

## Version of Library

## Expected Behaviour
When I call IDevice.PairingRequest() I expect a paring dialog to pop up that lets me pair with a peripheral. 

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Calling IDevice.PairingRequest() calls createBond() but no pop-up appears and the device does is not paired.

(Note: on Android L, IDevice.PairingRequest() pairs the peripheral without notifying the user).

Note: I currently have the following permissions listed in the AndroidManifest and I request them from MainActivity.OnCreate() if they are not already granted:

    Manifest.Permission.Bluetooth,
    Manifest.Permission.BluetoothAdmin,
    Manifest.Permission.AccessCoarseLocation,
    Manifest.Permission.AccessFineLocation,

This is what I see in the application output: 

    [BluetoothDevice] createBond() for device 80:7D:3A:86:DD:2A called by pid: 20620 tid: 20791

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-03-07:
Please leave a reproducible sample of some sort.  How are you calling the method?  Did you debug through the method?  Did you try it on the sample?  

It appears to me you are using this for professional use.  Please note that I'm not here for support.  OSS works if people contribute instead of just dropping a shopping list of things to look into.

Carbaholic, 2019-03-07:
It is certainly for professional use and I would love to contribute to the project. It is unfair to accuse me of "just dropping a shopping list of things to look into". 

With the last issue I posted I have not had the chance to debug it but I did post a solution for anyone else who has that issue. I also looked through your code and looked for any clear places where there could be a null reference exception and nothing jumped out at me or else I would have made a change and submitted a pull request. 

I appreciate all of your work but you are the saltiest library owner I have ever worked with.

aritchie, 2019-03-07:
And with that, find someone else to help you :)

closed 2019-03-07 by aritchie