aritchie/bluetoothle #299: Impossible to compile the project with latest Xamarin visual studio

ormaa, 2019-03-27:
Hello,

using Xamarin Visual studio latest version : 7.7.4 Build 1
got your repository
build to compile, and try to run on IOS Device.

I got this error : 
/Users/olivier/.nuget/packages/microsoft.build.tasks.git/1.0.0-beta-63127-02/build/Microsoft.Build.Tasks.Git.targets(5,5): Error MSB4018: The "Microsoft.Build.Tasks.Git.LocateRepository" task failed unexpectedly.
System.BadImageFormatException: Method has no body
File name: 'LibGit2Sharp'
  at LibGit2Sharp.Core.Proxy.git_buf_free (LibGit2Sharp.Core.Handles.GitBuf buf) [0x00000] in <06ab184b4ccd4259a7a7af2756f77645>:0 
  at LibGit2Sharp.Core.Handles.GitBuf.Dispose () [0x00000] in <06ab184b4ccd4259a7a7af2756f77645>:0 
  at LibGit2Sharp.Core.Proxy.ConvertPath (System.Func`2[T,TResult] pathRetriever) [0x0002e] in <06ab184b4ccd4259a7a7af2756f77645>:0 
  at LibGit2Sharp.Core.Proxy.git_repository_discover (LibGit2Sharp.Core.FilePath start_path) [0x00005] in <06ab184b4ccd4259a7a7af2756f77645>:0 
  at LibGit2Sharp.Repository.Discover (System.String startingPath) [0x00006] in <06ab184b4ccd4259a7a7af2756f77645>:0 
  at Microsoft.Build.Tasks.Git.GitOperations.LocateRepository (System.String directory) [0x00000] in /_/src/Microsoft.Build.Tasks.Git.Operations/GitOperations.cs:24 
  at Microsoft.Build.Tasks.Git.RepositoryTasks.LocateRepository (Microsoft.Build.Tasks.Git.LocateRepository task) [0x00000] in /_/src/Microsoft.Build.Tasks.Git.Operations/RepositoryTasks.cs:50 
  at Microsoft.Build.Tasks.Git.LocateRepository.Execute () [0x00038] in /_/src/Microsoft.Build.Tasks.Git/LocateRepository.cs:32 
  at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute () [0x00023] in /Users/builder/jenkins/workspace/build-package-osx-mono/2018-06/external/bockbuild/builds/msbuild-15/src/Build/BackEnd/TaskExecutionHost/TaskExecutionHost.cs:573 
  at Microsoft.Build.BackEnd.TaskBuilder.ExecuteInstantiatedTask (Microsoft.Build.BackEnd.ITaskExecutionHost taskExecutionHost, Microsoft.Build.BackEnd.Logging.TaskLoggingContext taskLoggingContext, Microsoft.Build.BackEnd.TaskHost taskHost, Microsoft.Build.BackEnd.ItemBucket bucket, Microsoft.Build.BackEnd.TaskExecutionMode howToExecuteTask) [0x001f6] in /Users/builder/jenkins/workspace/build-package-osx-mono/2018-06/external/bockbuild/builds/msbuild-15/src/Build/BackEnd/Components/RequestBuilder/TaskBuilder.cs:784  (MSB4018) (Plugin.BluetoothLE)

I tried to update the package, without success.

aritchie, 2019-03-27:
I'm not sure why you've filed this again when I've already answered that the library will not compile on VS4mac!  https://github.com/aritchie/bluetoothle/issues/258

closed 2019-03-27 by aritchie

ormaa, 2019-03-27:
Oh god.
Sorry about that,

I forgot that, and didn’t read all the item in bug area.

Sorry again.

aritchie, 2019-03-27:
No problems.  Thanks for responding.

Shortly after the release of VS2019 for Mac, this will be fixed anyhow.