aritchie/bluetoothle #300: Cannot read/write to characteristic when notifications enabled

lrolleman, 2019-04-04:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [check] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

Tested on Android 6 and 7

## Version of Library

6.2.3

## Expected Behaviour

After enabling notifications on a characteristic, I should still be able to read/write to that characteristic

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Attempting to read/write to a characteristic with enabled notifications causes the call to hang (and possibly throw TimeoutException if Timeout is set)

A workaround for this is to call DisableNotifications on the characteristic before the read/write, and EnableNotifications in the result + error handlers. Not ideal though. 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-04-04:
Please provide some code that I can work with here.

lrolleman, 2019-04-04:
I can't provide a link to the actual code but paraphrasing:

`characteristic.WhenNotificationsRecieved().Subscribe( result => { /* do something */} );`
`characteristic.EnableNotifications();`
`characteristic.Read().Subscribe(result => { /* do something */ });`

This hangs. 

If I do:
`characteristic.Read().Subscribe(result => { /* do something */ });`
`characteristic.WhenNotificationsRecieved().Subscribe( result => { /* do something */} );`
`characteristic.EnableNotifications();`

It works fine, at least until I try and read from that characteristic again.

aritchie, 2019-04-04:
I run code like this all of the time without issue, so I'm definitely going to need some sort of extra details.  

Can you define "hang", do you mean subscribe just isn't being called on Read or is that whole line blocking?  
Have you tried using the sample app with your host device?
Do you have an error catch in your subscribe?

lrolleman, 2019-04-04:
By hang I mean Read gets called but it never gets a result in Subscribe, it will just timeout after the specified interval (currently 30 seconds)
Not yet, will try that and report back
Yes, it just gets the TimeoutException

lrolleman, 2019-04-04:
The sample app works fine. I'm starting to think there might be something a bit more painful going on that just so happened to manifest itself between when I enabled the notifications and tried to read from the characteristics. 

When I add a 1 second delay between when I connect to the device and when I do the first read, all the problems seem to go away. This screams of a race condition somewhere, I'll investigate further and see if I can find anything useful

lrolleman, 2019-04-04:
Ok, I think I've found the culprit, for real this time. 

When I was connecting to the device, upon success I would immediately request an MTU size increase. However I didn't wait for the result (my thinking being I don't actually care when the size increases, since the next bunch of reads/writes are < 20 bytes anyway. Turns out this is bad thinking). 

This seems to really mess things up down the line. Perhaps a serial execution issue? Anyway, looks like false alarm on the notifications...

aritchie, 2019-04-04:
On android, it will definitely be an execution order issue.  I manage almost all read/writes with an internal mechanism to prevent things like this.  MTU is not under that guard, but I can fix it.  

An immediate workaround is to simply await the MTU request completion as you've said before moving to other commands.  All characteristic operations are already defended against this, so don't worry about them.

prajaktashindeSilicus, 2019-11-19:
@aritchie 
I have the following method.
public async Task RequestMTU(ScanResultViewModel selectedDevice)
        {
            if (!selectedDevice.Device.Features.HasFlag(DeviceFeatures.MtuRequests))
            {
                userDialogs.Alert("MTU Request not supported on this platform");
            }
            else
            {
                var actual = await selectedDevice.Device.RequestMtu(512);
                Console.WriteLine("MTU Changed to " + actual);
                //userDialogs.Toast("MTU Changed to " + actual);
            }
        }

  var actual = await selectedDevice.Device.RequestMtu(512); this line of code is awaited but I'm not getting any response further. My app gets hang. 

Please help me asap.