aritchie/bluetoothle #301: [Android] RefreshServices causes reads to hang

lrolleman, 2019-04-10:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

6.0 and 7.1

## Version of Library

6.2.3

## Expected Behaviour

When RefreshServices Android configuration is set to true, I should be able to read characteristics as normal

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

With RefreshServices enabled, I can only manage a few reads on different characteristics (which seem to progressively get slower) before they all just start timing out. 

Turning RefreshServices back off returns everything back to normal. 

I turned it on as i need to do OTA updates but having it always on seems to not play nicely when trying to do a large number of reads at once. 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-04-10:
Refresh services is only called when a discovery is requested.  Please include a full reproducible sample of what you are doing.  

https://github.com/aritchie/bluetoothle/blob/dev/Plugin.BluetoothLE/Platforms/Android/Device.cs#L102

lrolleman, 2019-04-10:
I added a test method to the sample app in DeviceViewModel like this
        
        private async void Test()
        {
            <relevant UUIDs defined here>
            for (var i = 0; i < 5; i++)
            {
                var writeChar = await device.GetKnownCharacteristics(serviceUuid, writeCharUuid);
                await writeChar.Write(new byte[] {1});
                var read1Char = await device.GetKnownCharacteristics(differentServiceUuid, read1CharUuid);
                var t1 = await read1Char.Read();
                var read2Char = await device.GetKnownCharacteristics(serviceUuid, read2CharUuid);
                var read3Char = await device.GetKnownCharacteristics(serviceUuid, read3CharUuid);

                var t2 = await read2Char.Read();
                var t3 = await read3Char.Read();
            }

and called from the WhenStatusChanged handler on connect.

I put a break point at the beginning and end of the loop. With RefreshServices enabled, it makes it through the loop once, hanging somewhere between the first and second read of read3Char. 

This works just fine with RefreshServices disabled. 

RefreshServices is enabled in MainActivity

Hope this helps