aritchie/bluetoothle #304: RegisterAndNotify problems after reconnection

Andufer, 2019-04-25:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 9

## Version of Library
6.2.3 Nuget

## Expected Behaviour

1. Discovery of the device
2. Connection to the device
3. The device connects
4. Subscription to a characteristic with RegisterAndNotify
5. The characteristic is subscribed and the callback is called when there is a new value
6. Disconnection of the device, from the device side (link loss, or device goes in sleep)
7. Device resumes from sleep or comes in range
8. Re-connection to the device, the WhenStatusChanged is called
9. Subscription to a characteristic with RegisterAndNotify
10. The characteristic is subscribed and the callback is called when there is a new value

## Actual Behavior

1. Discovery of the device
2. Connection to the device
3. The device connects
4. Subscription to a characteristic with RegisterAndNotify
5. The characteristic is subscribed and the callback is called when there is a new value
6. Disconnection of the device, from the device side (link loss, or device goes in sleep)
7. Device resumes from sleep or comes in range
8. Re-connection to the device, the WhenStatusChanged is called
9. Subscription to a characteristic with RegisterAndNotify
10. In my first case the characteristic is subscribed and the callback is called **_twice_** when there is a new value. In my second case the characteristics is subscribed, but the callback is not called when there's a new value.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

Hi,
I'm writing an App that communicate with an hardware via Bluetooth LE using Xamarin and Prism.
I'm using Plugin.BluetoothLE v6.2.3 on Visual studio 2019 version 16.02.
I've created a service with the scan method:

`
public void Scan()
        {
            scanner = CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    if (scanResult.Device.Name != null && scanResult.Device.Name.Contains("CromoMed"))
                    {
                        device = scanResult.Device;
                        scanner.Dispose();
                        scanner = null;

                        Pen.Current.Name = device.Name;
                        Pen.Current.ID = device.Uuid;

                        Thread.Sleep(1000); //Wait for stopping scanner

                        device.WhenStatusChanged().Subscribe(status =>
                        {
                            switch (status)
                            {
                                case ConnectionStatus.Connected:
                                    Pen.Current.IsConnected = true;
                                    break;
                                case ConnectionStatus.Disconnected:
                                    Pen.Current.IsConnected = false;
                                    break;
                            }
                        });

                        device.Connect();
                        GetCharacteristics();
                    }
                });
        }
`

GetCharacteristics is:

`
private void GetCharacteristics()
        {
            device.WhenAnyCharacteristicDiscovered().Subscribe(ch =>
            {
                if (ch.Uuid == penRx)
                {
                    penWrite = ch;
                    Debug.WriteLine("*********** Characteristics Write found *********");
                }
                if (ch.Uuid == penTx)
                {
                    penRead = ch;
                    Debug.WriteLine("*********** Characteristics Read found *********");

                    penRead.RegisterAndNotify(true, true).Subscribe(res =>
                    {
                        Read(res.Data);
                        Debug.WriteLine("*********** Data received *********");
                    });
                }
            });
        }
`

When I debug the application on a Samsung S9+, all is work properly, but if I try to turn off my hardware and then turn on again, the Read process send 2 notifications instead of 1. If I turn off and on the hardware one more time, the Read process send 3 notifications.
I've tryied with this new version of GetCharacteristics:

`
        private void GetCharacteristics()
        {
            bool writeDiscovered = false;
            bool readDiscovered = false;
            bool readRegistered = false;

            device.WhenAnyCharacteristicDiscovered().Subscribe(ch =>
            {
                if (ch.Uuid == penRx && !writeDiscovered)
                {
                    writeDiscovered = true;
                    penWrite = ch;
                    Debug.WriteLine("*********** Characteristics Write found *********");
                }
                if (ch.Uuid == penTx && !readDiscovered)
                {
                    readDiscovered = true;
                    penRead = ch;
                    Debug.WriteLine("*********** Characteristics Read found *********");

                    penRead.RegisterAndNotify(false, false).Subscribe(res =>
                    {
                        Read(res.Data);
                        Debug.WriteLine("*********** Data received *********");
                    });
                }
            });
        }
`

But, in this case, when I turn off and on the hardware, the Read does not send any notification.

Where I'm wrong?

Thanks a lot and have a nice day,
Andrea

pravincarvalho, 2019-04-25:
Hi Andrea,

You need to enable notifications each time you reconnect. If you are also going to register a new subscription each time you connect you could dispose off the subscription when a disconnect happens.

See an example of .DisposeWith in the Sample code.

https://github.com/aritchie/bluetoothle/blob/eddbe927036263179ba4079318bff99524c2da94/Samples/Samples/DeviceViewModel.cs#L116-L120

Andufer, 2019-04-25:
Thanks a lot,
I'm trying to understand the example.
But I don't understand why in my first case of using RegisterAndNotify, the reconnection is OK and the problem is that the notification is twice. Boh!

Start studying the example now!!!
Thanks a lot and have a nice day,
Andrea

Andufer, 2019-04-25:
Hi pravincarvalho,
I'm trying to write a new class as in the example you suggest, but after I defined my device as:
IDevice device;
the ObserveOn and DisposeWith do not exists. I just have WhenReadRssContinuosly and Subscribe. 
Maybe from 2018 to now somthing is changed?

pravincarvalho, 2019-04-25:
You need https://reactiveui.net/ for those Observable methods. If you don't want those extension methods you could also just call Dispose on the subscription

````
subscription = penRead.RegisterAndNotify(true, true).Subscribe(res =>
                {
                    Read(res.Data);
                    Debug.WriteLine("*********** Data received *********");
                });
// On disconnect
subscription.Dispose();

aritchie, 2019-04-25:
Usually, duplicate actions happen because you hang on to a reference of characteristic somewhere.  Try not to do this.  There are other extensions in the library that can help.  

There is also an IsNotify flag on the characteristic.  You should check its value.

Andufer, 2019-04-25:
Thanks aritchie,
I've tried to use Dispose of the subscription and also check the IsNotify, but instead of receiving two notification, I don't receive any notification.
I've added the following  method:
`
private void RemoveCharacteristics()
        {
            if (readSub != null)
                readSub.Dispose();
            readSub = null;
            penRead = null;
            penWrite = null;
            Debug.WriteLine("*********** Characteristics removed *********");
        }
`
and added the call when the connection go to disconnected:
`
public void Scan()
        {
            scanner = CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    if (scanResult.Device.Name != null && scanResult.Device.Name.Contains("CromoMed"))
                    {
                        device = scanResult.Device;
                        scanner.Dispose();
                        scanner = null;

                        Pen.Current.Name = device.Name;
                        Pen.Current.ID = device.Uuid;

                        Thread.Sleep(1000); //Wait for stopping scanner

                        device.WhenStatusChanged().Subscribe(status =>
                        {
                            switch (status)
                            {
                                case ConnectionStatus.Connected:
                                    Pen.Current.IsConnected = true;
                                    break;
                                case ConnectionStatus.Disconnected:
                                    Pen.Current.IsConnected = false;
                                    RemoveCharacteristics();
                                    break;
                            }
                        });

                        device.Connect();
                        GetCharacteristics();
                    }
                });
        }
`
GetCharacteristic is now:

`
private void GetCharacteristics()
        {
            bool writeDiscovered = false;
            bool readDiscovered = false;

            device.WhenAnyCharacteristicDiscovered().Subscribe(ch =>
            {
                if (ch.Uuid == penRx && !writeDiscovered)
                {
                    penWrite = ch;
                    Debug.WriteLine("*********** Characteristics Write found *********");
                }
                if (ch.Uuid == penTx && !readDiscovered)
                {
                    penRead = ch;
                    Debug.WriteLine("*********** Characteristics Read found *********");

                    readSub = penRead.RegisterAndNotify(true, true).Subscribe(res =>
                    {
                        Read(res.Data);
                        Debug.WriteLine("*********** Data received *********");
                    });
                }
            });
        }
`

Just adding the readSub.Dispose(), and I don't receive any notification after turning off and on the hardware.
Where is the problem?
Can you tell me what extension I've to use if instead of RegisterAndNotify?
If you need I can send you the project, but I know you are very busy and I don't want to abuse your time.

Thanks in any case and have a great day.
Andrea

Andufer, 2019-04-25:
Sorry, I've also tried to use the flag IsNotifying, but the result is the same. No notification after disconnec/reconnect.
`
private void GetCharacteristics()
        {
            bool writeDiscovered = false;
            bool readDiscovered = false;

            device.WhenAnyCharacteristicDiscovered().Subscribe(ch =>
            {
                if (ch.Uuid == penRx && !writeDiscovered)
                {
                    penWrite = ch;
                    Debug.WriteLine("*********** Characteristics Write found *********");
                }
                if (ch.Uuid == penTx && !readDiscovered)
                {
                    penRead = ch;
                    Debug.WriteLine("*********** Characteristics Read found *********");

                    if (!penRead.IsNotifying)
                        readSub = penRead.RegisterAndNotify(true, true).Subscribe(res =>
                        {
                            Read(res.Data);
                            Debug.WriteLine("*********** Data received *********");
                        });
                }
            });
        }
`

aritchie, 2019-04-28:
Is your host device hanging on to subscriptions?

Andufer, 2019-04-29:
When I turn off my device, I turn off all the timers, so It can not hang on to the subscriptions.

Andufer, 2019-05-03:
Hi Allan,
I’ve not received any answer on my last post.
Do you have any news?

Thanks and have a nice day,
Andrea

aritchie, 2019-05-15:
So I've tested this scenario on my local devices and all appears to be working fine for me in the sample and other apps I have.  You'll need to provide a full (but reasonably small) reproducible sample if you want assistance on your issue.

I would also suggest checking the host device (not your mobile app)

closed 2019-05-19 by aritchie