aritchie/bluetoothle #306: Can not discover any device

bastianbecker21, 2019-05-03:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] Android

## Version of OS(s) listed above with issue
Android 6.0 & 7.0

## Version of Library
6.2.3

## Expected Behaviour
Discover my devices that I enabled Bluetooth. One is android phone, another one is bluetooth speaker

## Actual Behavior
Discovered nothing

## Code of Conduct
// discover some devices
CrossBleAdapter.Current.Scan().Subscribe(scanResult => {});

aritchie, 2019-05-03:
You need to give a lot more details than this if you want help!! 

Right now, my guess is that your android device isn’t advertising and your Bluetooth speaker isn’t Bluetooth LE at all since no there isn’t any Bluetooth speakers on the market that are BLE

bastianbecker21, 2019-05-03:
I am a beginner in bluetooth technology so my comment is not enough to explain about my issue exactly. I got a point from your point, My bluethooth speaker has bluetooth 4.2 but may not be a BLE as you said.
and I am not sure about advertising of my device, How can I make it advertising? thanks for your help

aritchie, 2019-05-03:
The question goes a bit beyond the nature of a github issue.  You can use the GATT server located in the library to create your own advertising with your mobile device, but for that, I would suggest using the samples and docs to move forward.

Best of luck

closed 2019-05-03 by aritchie