aritchie/bluetoothle #307: WatchOS Support

mostlyjason, 2019-05-12:
I'd like to make a request for WatchOS support. I'd be willing to buy you many coffees or potentially contribute. I'm looking to add a Bluetooth central on my watch to connect to my desktop. I'm not sure whether upgrading this library will be easier or just writing my own iOS code in Xamarin.

aritchie, 2019-05-12:
I'll see what's involved.  Last time I checked, the API surface on watchos was "deficient" for a good portion of the feature set of this library

mostlyjason, 2019-05-12:
Yes it has limited features. This [tutorial](https://www.raywenderlich.com/336-core-bluetooth-in-watchos-tutorial) points out several limitations:

> watchOS and tvOS both rely on Bluetooth as their main system input, so Core Bluetooth has restrictions, to ensure system activities can run. Both can be only the central device, and can use at most two peripherals at a time. Peripherals are disconnected when the app is suspended.And the minimum interval between connections is 30ms, instead of 15ms for iOS and macOS.

aritchie, 2019-05-13:
This isn’t something I want to take on.  Too many missing function points.  Sorry.  Best of luck

closed 2019-05-13 by aritchie