aritchie/bluetoothle #308: Gatt

mmkrishna2008, 2019-05-13:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - 

- [ ]  Android

 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android Oreo 8.0

## Version of Library
6.2.3

## Expected Behaviour
I have implemented GATT server with a custom BLE Service and a single Notify/Read characteristic. After the client subscribe for a notification,  the server is able to push the data via notifications for sometime.  But the client is getting GATT error 133 on onDescriptorWrite callback.  It seems the GATT server is not returning any response for GATT Descriptor write request.

## Actual Behavior
The GATT server should return a write response to client when the client subscribe for a notification (onDescriptorWrite callback in client). 
``
this.gattServer = await this.adapter.CreateGattServer();
                var service = this.gattServer.CreateService(Service_Guid, true);
                this.dataCharacterstic = service.AddCharacteristic(Char_Guid, CharacteristicProperties.Notify | CharacteristicProperties.Read,
                     GattPermissions.Read | GattPermissions.Write);
                this.gattServer.AddService(service);
//Subscribe for Callbacks
                this.dataCharacterstic
                    .WhenDeviceSubscriptionChanged()
                    .Subscribe(e =>
                    {
//Code to update UI based on the subscription result
}
``
## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-05-13:
The error is on the client - not the server.  I can’t help you here.  GATT133 on Android usually means you are doing something in an asynchronous fashion before waiting for commands to complete.

mmkrishna2008, 2019-05-13:
@aritchie Thanks for quick reply. But i am facing the same error while testing the server with Nordic nrf Connect app as well. Can you please let me know the library class where the server is sending response for notification subscription request?

aritchie, 2019-05-13:
Can you share some sort of stack trace please.  I have no idea where the error is right now

closed 2019-05-17 by aritchie

mmkrishna2008, 2019-05-21:
@aritchie It looks like the mGattServer.sendResponse method needs to be called when notifications enabled on a characteristic. Can you please confirm this from your end?

Sample Android GATT code with this method implementation is working fine. 
I don't see responseNeeded implemented in this class.
https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE/Platforms/Android/Server/Internals/GattServerCallbacks.cs

Which method needs to be implemented in this library to send the response to GATT clients?

@Override
        public void onDescriptorWriteRequest(BluetoothDevice device, int requestId, 
BluetoothGattDescriptor descriptor, 
                        boolean preparedWrite, boolean responseNeeded, int offset, byte[] value) {
            Log.d("HELLO", "Our gatt server descriptor was written.");
            super.onDescriptorWriteRequest(device, requestId, descriptor, preparedWrite, responseNeeded, offset, value);
            Log.d("DONE", "Our gatt server descriptor was written.");

            //NOTE: Its important to send response. It expects response else client will disconnect
            if (responseNeeded) {
                mGattServer.sendResponse(device,
                        requestId,
                        BluetoothGatt.GATT_SUCCESS,
                        0,
                        value);

            }
        }