aritchie/bluetoothle #311: iOS charactisitcs not found

nosnikta10, 2019-05-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
All

[Characteristics.zip](https://github.com/aritchie/bluetoothle/files/3205069/Characteristics.zip)

## Version of Library
current

## Expected Behaviour
Discover characteristics should find all characteristics available on the service 

## Actual Behavior
It is possible for characteristics to share a common guid, but have different properties. The ios implementation of discover characteristics is using a dictionary which is used to store found characteristics, and prevent duplicate characteristics from being returned. Using the guid to determine if the characteristic has already been found is not sufficient, it is necessary to also  verify the properties for the characteristic also match.

 if (!characteristics.ContainsKey(ch.Uuid))
   {
        characteristics.Add(ch.Uuid, ch);
        ob.OnNext(ch);
  }

the dictionary could be replaced with a KeyValuePair and an extension method could be used to determine if the characteristic should be added.

if (!characteristics.ExistsWithProperty(ch))
{                     
  characteristics.Add(new KeyValuePair<Guid, IGattCharacteristic>(ch.Uuid, ch));
   ob.OnNext(ch);
}  

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

MKGNZ, 2019-09-11:
@nosnikta10  I'm baffled with this. "characteristics to share a common guid". 
Do you have an example of such UUID?

There is only one API to get a characteristic. GetKnownCharacteristics(<uuid>); 
If you pass a single UUID you will get that single Characteristic. The same with Apple documentation:
func discoverCharacteristics([CBUUID]?, for: CBService) [https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518797-discovercharacteristics](url)

Yes a characteristic might have more Descriptors or attributes, If it is write, notify, read, indicate, .....
But still is only one Characteristic. There is no API (iOS, Android, ...) to get a characteristic with the same UUID and different properties.

nosnikta10, 2019-09-19:
Hi aritchie,

Sorry I didn't describe the issue correctly. You are correct that there can only be a single characteristic, but a characteristic can have different atts/descriptors. The issue is when a characteristic has different  atts/descriptors the ios implementation would only get the first and skip any other variations due to using a dictionary based on the characteristics guid.

As mentioned above, I simply changed this to a key value pair and all variations are found.

Regards,
Cam

MKGNZ, 2019-09-20:
Oh now I get it **BUT** still you get the One Characteristic and then you call:
DiscoverDescriptors();

Like:
```
                _Service.DiscoverCharacteristics()
                    .Subscribe(ch => 
                    {
                        // Do whatever you need to do with your Characteristic
                        // .....
                        // and then:
                        ch.DiscoverDescriptors().Subscribe(d =>
                        {
                            // Do something with the descriptor
                            if (d.Description == "this and that")
                                DoFoo();
                            
                            if (d.Uuid == GuidYouLookingFor)
                                DoAnotherFoo();
                        });
                    });

```

nosnikta10, 2019-09-20:
Hey Archie, 

Yes I understand you could do it this way, but to bring the ios inline with the android implementation I had to modify the ios discover characteristics to return each variation of the characteristic. The android implementation returns all the different varations e.g.... read/write/notify when subscribing to discover characteristics.

Cheers