aritchie/bluetoothle #312: System.MissingMethodException: Method not found: Plugin.BluetoothLE.IAdvertisementData

syedadeel2, 2019-05-26:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS
Android

## Version of Library
6.2.3

## Expected Behaviour
```
                 CrossBleAdapter.Current.Scan().Subscribe((x) =>
                    {
                        if (x.AdvertisementData != null)
                        {
                            if (x.AdvertisementData.IsConnectable)
                            {
                                callback(x);
                            }
                        }

                    });
```
Expecting to return the AdvertisementData but throwing exception here see the video in action
http://somup.com/cqhOc3n5ZM

## Actual Behavior
```
Unhandled Exception:
System.MissingMethodException: Method not found: Plugin.BluetoothLE.IAdvertisementData Plugin.BluetoothLE.ScanResult.get_AdvertisementData() occurred
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* If you have a complaint about the docs, FIX IT - DO NOT FILE IT HERE!  I won't guarantee everything is always 100% up-to-date since I am just 1 person.
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2019-05-26:
What platform?  
Are you using packages.config?  
If android - are you targeting android 8.1+
Please provide a reproducible sample

syedadeel2, 2019-05-26:
Issue Resolved, having Conflict with other BLE plugins. I have clean my project with useless references and all good now. 

Thank you for your time looking forward to have Serial style communication in this plugin. Rather than RX.

closed 2019-05-26 by syedadeel2

aritchie, 2019-05-26:
This plugin will always be RX so you’ll likely need to look elsewhere