aritchie/bluetoothle #313: BookmarkBluetooth LE Notify Issue in Xamarin.Forms

mallikarjunpatala, 2019-06-07:
Application and my Hardware. Please help if anyone have come across the same issue.

using PCLCrypto;
using Plugin.BluetoothLE;
using Plugin.BluetoothLE.Server;
using System;
using System.Collections.Generic;
using System.Diagnostics;
using System.Reactive.Linq;
using System.Security.Cryptography;
using System.Text;
using Xamarin.Forms;
using Xamarin.Forms.Xaml;

namespace PEPS
{
[XamlCompilation(XamlCompilationOptions.Compile)]
public partial class Register : ContentPage
{
//Registration
string REGISTER_SERVICE_UUID = "4d7852c8-6fd8-11e9-a923-1681be663d3e";
string REGISTER_CHARACTERISTIC_UUID = "4d7852c9-6fd8-11e9-a923-1681be663d3e"; //notify
string CALIBRATE_CHARACTERISTIC_UUID = "4d7852ca-6fd8-11e9-a923-1681be663d3e"; //Read

 //Localization
    string LOC_SERVICE_UUID = "4d7855a2-6fd8-11e9-a923-1681be663d3e";
    string LOC_CHARACTERISTIC_UUID = "4d7855a3-6fd8-11e9-a923-1681be663d3e";    //Write 
    string RKE_CHARACTERISTIC_UUID = "4d7855a4-6fd8-11e9-a923-1681be663d3e";    //Read

    public Register()
    {
        InitializeComponent();
    }

    //ToolBarRefresh

    public async void Syncpage(object sender, EventArgs e)
    {
        await Navigation.PushAsync(new MainPage());
    }

    void OnEvent(string msg) => Device.BeginInvokeOnMainThread(() =>
       lblReadData.Text += msg + Environment.NewLine + Environment.NewLine
   );
    public async void CommonButtonForRegister(object sender, EventArgs e)
    {
        Button btn = (Button)sender;
        var classId = btn.ClassId;

        switch (classId)
        {
            case "btn_connect":                 
                try
                {
                    //Gatt Server creation
                    var gattServer = await CrossBleAdapter.Current.CreateGattServer();

                    // Register Service Creation
                    var reg_service = gattServer.CreateService(Guid.Parse(REGISTER_SERVICE_UUID), true);
                    var reg_notifyCharacteristic = reg_service.AddCharacteristic(Guid.Parse(REGISTER_CHARACTERISTIC_UUID),                                  CharacteristicProperties.Notify, GattPermissions.Read);
                    var reg_readCharacteristic = reg_service.AddCharacteristic(Guid.Parse(CALIBRATE_CHARACTERISTIC_UUID), 
                    CharacteristicProperties.Read, GattPermissions.Read);
                    gattServer.AddService(reg_service);

                    // Notify Event
                    IDisposable notifyBroadcast = null;
                    reg_notifyCharacteristic.WhenReadReceived().Subscribe(d =>
                        {

                        });
                    reg_notifyCharacteristic.WhenWriteReceived().Subscribe(d =>
                    {

                    });
                    reg_notifyCharacteristic.WhenDeviceSubscriptionChanged().Subscribe(d =>
                    {
                        var @event = d.IsSubscribed ? "Subscribed" : "Unsubcribed";
                        this.OnEvent($"Device {d.Device.Uuid} {@event}");
                        this.OnEvent($"Charcteristic Subcribers: {reg_notifyCharacteristic.SubscribedDevices.Count}");

                        //var isSubscribed = d.IsSubscribed ? "Subscribed" : "Unsubcribed";
                        if (notifyBroadcast == null)
                        {
                            notifyBroadcast = Observable
                                            .Interval(TimeSpan.FromSeconds(1))
                                            .Where(x => reg_notifyCharacteristic.SubscribedDevices.Count > 0)
                                            .Subscribe(a =>
                                            {
                                                //var isSubscribed = d.IsSubscribed ? "Subscribed" : "Unsubcribed";

                                                //Debug.WriteLine("Sending Broadcast");
                                                //var enData = Crypto.CreateSalt(16);
                                                //byte[] arr1 = { 0x00, 0x01, 0x00, 0x00 };
                                                //var bytes = EncryptAes(arr1, enData);
                                                //reg_notifyCharacteristic.Broadcast(bytes);
                                                //var str = Crypto.DecryptAes(bytes, "MySecretKey", enData);
                                            });
                        }
                    });

                    // Read Operation
                    reg_readCharacteristic.WhenReadReceived().Subscribe(x =>
                    {
                        var read = "PEPS";
                        x.Value = Encoding.UTF8.GetBytes(read);
                        x.Status = GattStatus.Success;
                    });

                    //Advertising the Packet
                    var guid1 = new Guid("000055A2-0000-1000-8000-00805F9B34FB");
                    var guid2 = new Guid("000052C8-0000-1000-8000-00805F9B34FB");

                    CrossBleAdapter.Current.Advertiser.Start(
                    new AdvertisementData
                    {
                        LocalName = "PEPS Device",
                        ServiceUuids = new List<Guid> { guid1, guid2 }
                    });
                    lblConnected.Text = "Connection Established Successfully";

                    // Localisation Service Creation
                    var loc_service = gattServer.CreateService(Guid.Parse(LOC_SERVICE_UUID), true);
                    var loc_writeCharacteristic = loc_service.AddCharacteristic(Guid.Parse(LOC_CHARACTERISTIC_UUID),        CharacteristicProperties.Write, GattPermissions.Write);
                    loc_service.AddCharacteristic(Guid.Parse(RKE_CHARACTERISTIC_UUID), CharacteristicProperties.Read, GattPermissions.Read);
                    gattServer.AddService(loc_service);

                    //Write Operation
                    loc_writeCharacteristic.WhenWriteReceived().Subscribe(x =>
                    {
                        string write = string.Empty;
                        try
                        {
                            write = Encoding.UTF8.GetString(x.Value, 0, x.Value.Length);
                            this.OnEvent($"Characteristic Write Received - {write}");
                        }
                        catch (Exception ex)
                        {
                            lblReadData.Text = ex.Message;
                        }
                    });
                }
                catch (Exception ex)
                {
                    var mesage = ex.Message;
                    await DisplayAlert("Warning!", ex.Message, "Ok");
                }
                break;/*
        }
    }
}

aritchie, 2019-06-07:
What exactly is the issue?  What OS is this on?

Please fill out the issue template in the future!!

mallikarjunpatala, 2019-06-10:
Android version : Orio, IOS-10, unable to establish notification from server to client using CharacteristicProperties.Notify. For that we used function "notifyCharacteristic.WhenDeviceSubscriptionChanged().Subscribe(d =>{   }

1. In android it is contineously loading and getting disconnected and showing the error 
"Error 133(0x85): GATT [ERROR,]
"Error 22(0x16): GATT CONN TERMINATE LOCAL HOST 

Note : Please find this video for your reference. In that when we clicked on notify, the progress ring is contineously loading and at certain point it is getting disconnected by giving error.

In Android Manifest we are using these permissions:
	<uses-permission android:name="android.permission.BLUETOOTH"  />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />

Please do let us know if we miss something.

[20190610_151744.zip](https://github.com/aritchie/bluetoothle/files/3271594/20190610_151744.zip)

Please kindly help us.
Thanks in Advance.

aritchie, 2019-06-11:
What type of phone is this hosted on? 
Is the device connecting paired to your host?  If so, have you unpaired it?

mallikarjunpatala, 2019-06-12:
We are using Samsung S8 mobile and version is Oreo with all type of devices (RedMI, Asus, Samsung S8) all with Oreo version.

We checked with paired and unpaired also we tested. 

We have given these permissions in Android manifest

<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
	<uses-permission android:name="android.permission.LOCATION_HARDWARE" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.READ" />
	<uses-permission android:name="com.android.voicemail.permission.READ_VOICEMAIL" />
	<uses-permission android:name="android.permission.BROADCAST_STICKY" />
	<uses-permission android:name="android.permission.BROADCAST_WAP_PUSH" />
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

mallikarjunpatala, 2019-06-12:
We used these permissions

[image]
[image]

mmkrishna2008, 2019-06-14:
GATT Client expects a response from server when enabling notification on a characteristic. As this library is not sending the response in onDescriptorWriteRequest method the client will get 133 error onDescriptorWriteResult method and will disconnect the connection.

bluetoothle/Plugin.BluetoothLE/Platforms/Android/Server/GattCharacteristic.cs (The code in bold will fix this disconnection issue)

`IObservable<DeviceSubscriptionEvent> subscriptionOb;
        public override IObservable<DeviceSubscriptionEvent> WhenDeviceSubscriptionChanged()
        {
            this.subscriptionOb = this.subscriptionOb ?? Observable.Create<DeviceSubscriptionEvent>(ob =>
            {
                var handler = new EventHandler<DescriptorWriteEventArgs>((sender, args) =>
                {
                    if (args.Descriptor.Equals(this.NotificationDescriptor))
                    {
**System.Diagnostics.Debug.WriteLine("Args.ResponseNeeded - {args.ResponseNeeded}");
                        if (args.ResponseNeeded)
                        {
                            lock (this.context.ServerReadWriteLock)
                            {
                                this.context.Server.SendResponse
                                (
                                    args.Device,
                                    args.RequestId,
                                    ProfileState.Connected, //GattStatus.Success , but VS is throwing error when using GattStatus enum in this method. 
                                    args.Offset,
                                    args.Value
                                );
                            }
                        }**
                        if (args.Value.SequenceEqual(NotifyEnabledBytes) || args.Value.SequenceEqual(IndicateEnableBytes))
                        {
                            var device = this.GetOrAdd(args.Device);
                            ob.OnNext(new DeviceSubscriptionEvent(device, true));
                        }
                        else
                        {
                            var device = this.Remove(args.Device);
                            if (device != null)
                                ob.OnNext(new DeviceSubscriptionEvent(device, false));
                        }
                    }
                });
                var dhandler = new EventHandler<ConnectionStateChangeEventArgs>((sender, args) =>
                {
                    if (args.NewState != ProfileState.Disconnected)
                        return;

                    var device = this.Remove(args.Device);
                    if (device != null)
                        ob.OnNext(new DeviceSubscriptionEvent(device, false));
                });

                this.context.Callbacks.ConnectionStateChanged += dhandler;
                this.context.Callbacks.DescriptorWrite += handler;

                return () =>
                {
                    this.context.Callbacks.DescriptorWrite -= handler;
                    this.context.Callbacks.ConnectionStateChanged -= dhandler;
                };
            })
            .Publish()
            .RefCount();

            return this.subscriptionOb;
        }
`

commit 2c168979 (2019-06-19): fix #313 - writeback when response needed for notification descriptor
closed 2019-06-19 by aritchie