aritchie/bluetoothle #314: Unable to advertise non connectable packets in Plugin.BLE

mallikarjunpatala, 2019-06-17:
W e are unable to advertise non connectable packets using Plugin.BLE.

Even set the flag in advertiser 

 "  var settings = new AdvertiseSettings.Builder()
                .SetAdvertiseMode(AdvertiseMode.Balanced)
                .SetConnectable(false);  "  to " false " also not able to make it non connectable.

PFA. In that after making it as non connectable it is showing "Conneectable : Yes" and it is showing the "connect" button beside the device.

Actually result:  it should not show as "Conneectable : Yes" and "Connect"
[image]
 button

Is there any flags that we need to change along with this.

Please kindly help us to resolve this issue.
@aritchie 
Please share your mobile number will be in touch with you if not please share mail address. 
[image]
[image]

aritchie, 2019-06-17:
@mallikarjunpatala Please note that this is not technical support forums.  I also only give out direct lines of communication for paid engagements.  If you look at the AdvertisementData in the StartAdvertising call, there is nothing that allows you to control that flag which is intentional since iOS doesn't let you control it.

aritchie, 2019-06-18:
This has been answered.  Please don’t send multiple communications on other channels.  If the library doesn’t do what you need it to do, please submit aPR.  this isn’t a bug.  It just isn’t something I implement

closed 2019-06-18 by aritchie