aritchie/bluetoothle #318: [Bug]

ysavit, 2019-07-09:
- [x ] iOS
 - [ x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.5
## Version of OS(s) listed above with issue

11.00

## Steps to Reproduce
_knownDevices = CrossBleAdapter.Current.GetKnownDevice(Guid.Parse(uuid));
                    _knownDevices.Subscribe(device =>
                    {
                        //ConnectDevice(device);
                        device.WhenConnected().Subscribe((d) =>
                        {
                            if (!IsSubscribed)
                            {
                                d.WhenAnyCharacteristicDiscovered().Subscribe(charact =>
                                    {
                                           if (charact.Service.Uuid == Guid.Parse("<service uuid>"))
                                              {
                                                   if (charact.CanNotify() && !charact.IsNotifying)
                                                     {
                                                           charact.RegisterAndNotify().Subscribe(result =>
                                                           {
                                                           var data = (int)result.Data[0];
                                                                   if (IsTestMode)
                                                                         ProcessTestNotification(data);
                                                                   else
                                                                         ProcessNotification(data);
                                                            });
                                                      IsSubscribed = true;
                                               }
                                           }
                                    });
                            }
                        });
                        device.Connect(new ConnectionConfig { AutoConnect = true });
                    });
                }

## Expected Behavior
Character Notification subscribed

## Actual Behavior
failed on d.WhenAnyCharacteristicDiscovered().Subscribe
{System.NullReferenceException: Object reference not set to an instance of an object.
  at Plugin.BluetoothLE.Device.<DiscoverServices>b__16_0 (System.IObserver`1[T] ob) [0x00051] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\Platforms\Android\Device.cs:120 
  at System.Reactive.Linq.QueryLanguage+CreateWithDisposableObservable`1[TSource].SubscribeCore (System.IObserver`1[T] observer) [0x00000] in <269857168607470d991a5a72ee3bb154>:0 
  at System.Reactive.ObservableBase`1[T].Subscribe (System.IObserver`1[T] observer) [0x00037] in <269857168607470d991a5a72ee3bb154>:0

aritchie, 2019-07-11:
You listed you have this bug on iOS as well?  Is there a stack trace there as well?

aritchie, 2019-07-11:
Also, what version of Android and what type of device does this issue appear?

aritchie, 2019-07-20:
no follow up response

closed 2019-07-20 by aritchie