aritchie/bluetoothle #319: Upgrading to 6.2.5 - white screen on app startup

bodeg, 2019-07-09:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.5

## Version of OS(s) listed above with issue
Android 9.0

## Steps to Reproduce
My application works fine with Plugin.BluetoothLE 6.2.3 
After upgrading to 6.2.5 - the application starts and stuck on white screen.

No exceptions in debug log.
Same result when upgrading to 6.2.4 

I put a break point in my object constructor that deals with all the BluetoothLE.
in the constructor - the application stuck in the following (first) line:
`
_adapter = CrossBleAdapter.Current;
`

I changed my main page to a simple page - with a button - on button pressed I navigate to the original main page.
The application starts, no white page - but once I navigate to the original main page  - I am getting  the same behavior.
 
**Any idea what I am doing wrong?**

## Expected Behavior

## Actual Behavior
Once I downgrade to 6.2.3 everything works fine.

aritchie, 2019-07-09:
You aren’t really giving me anything to go on.  I’m going to guess you aren’t setting you Android target to 8.1+

bodeg, 2019-07-09:
Thanks
The target is set to 8.1

When the application starts I find the following error in the output
window, I am not sure it is related.

mono_pe_file_map: Error opening file Mono.Android.dll (3): No such file or
directoryLoaded assembly:
/storage/emulated/0/Android/data/com.step_hear.sh.app/files/.__override__/Acr.Core.dll
[External]07-09 20:01:31.390 I/tep_hear.sh.ap(22381): Late-enabling
-Xcheck:jni

When I upgrade to 6.2.5 - system.reactive was upgraded as well
If one of the other plugins is using older system.reactive - do you think
it might cause a conflict?

I continue exploring...and let you know

aritchie, 2019-07-11:
Please send a reproducible sample.  I don't just want to start throwing around guesses.

hudaIT, 2019-07-12:
I had the same issue it said implementation failed or something like that

bravekit, 2019-07-17:
I had the same error in my Android project. But for me, it was fixed with 6.2.6 upgrade

aritchie, 2019-07-18:
There is a patch in 6.2.6 that should fix this.  Since I never got a reply to my questions, I'm closing this as resolved

closed 2019-07-18 by aritchie

hudaIT, 2019-07-19:
Yes thank you very much its solved in 6.2.6