aritchie/bluetoothle #323: [Bug]

mmkrishna2008, 2019-07-20:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ -] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue

Android 8.0

## Steps to Reproduce
1. Start GATT Server using sample app and connect to GATT Server using NRF Connect mobile app as GATT Client
2. Enable Notifications on characteristic from NRF Connect app
3. Stop GATT Server (GattServer.Dispose())
4. GATT Client is not getting disconnected

## Expected Behavior
Stopping GATT Server would inform the client to disconnect.  I tried to call GattServer.ClearServices or GattServer.RemoveService before calling GattServer.Dispose method but no luck.

The same implementation is working fine in iOS. The client app is getting disconnected once the GATT Server is stopped by calling Dispose method.

I am not sure its an issue with Android or library. I will test this with native Android GATT Server implementation and let you know my findings.

## Actual Behavior
Stopping the GATT Server will notify the client to disconnect

mmkrishna2008, 2019-07-20:
Just a quick update. I have tested Google Sample GATT server app and found that Gatt Client(nRF Connect mobile app) is not getting disconnected after calling GattServer.Close() method. After killing the GATT Server , the advertisement got stopped but somehow android client is still maintaining the previous connection.

I thought of trying a workaround by informing the client app to disconnect (via notification) before stopping the GATT Server. 
Any workaround or proper solution to fix this issue?

aritchie, 2019-07-20:
this is simply the way android behaves and therefore not an issue with the library itself.

closed 2019-07-20 by aritchie