aritchie/bluetoothle #324: scanResult.AdvertisementData?.ServiceUuids throws exception

bodeg, 2019-07-21:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Feature Library

 - [ ] Core
 - [ X] BluetoothLE
 - [ ] Locations
 - [ ] HTTP Transfers
 - [ ] Notifications
 - [ ] Sensors

## Version of Library
Plugin.BluetoothLE 6.2.6

## Version of OS(s) listed above with issue
Android Pie 

## Steps to Reproduce
for some BLE devices  the following line of code throws System.ArgumentException
`
var serviceUuids = scanResult.AdvertisementData?.ServiceUuids;
`

## Expected Behavior
I don`t own those devices - so I am not sure what should be the result (?)

## Actual Behavior

Exception details:

 System.ArgumentException: length
   at System.Array.Copy (System.Array sourceArray, System.Int32 sourceIndex, System.Array destinationArray, System.Int32 destinationIndex, System.Int32 length) [0x000c3] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
   at Plugin.BluetoothLE.Internals.InternalScanRecord.Parse (System.Byte[] scanRecord) [0x00037] in C:\dev\bluetoothle\Plugin.BluetoothLE\Platforms\Android\Internals\InternalScanRecord.cs:27 
   at Plugin.BluetoothLE.AdvertisementData+<>c__DisplayClass2_0.<.ctor>b__0 () [0x00000] in C:\dev\bluetoothle\Plugin.BluetoothLE\Platforms\Android\AdvertisementData.cs:22 
   at System.Lazy1[T].ViaFactory (System.Threading.LazyThreadSafetyMode mode) [0x00043] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
   at System.Lazy1[T].ExecutionAndPublication (System.LazyHelper executionAndPublication, System.Boolean useDefaultConstructor) [0x00022] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
   at System.Lazy1[T].CreateValue () [0x00074] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
   at System.Lazy1[T].get_Value () [0x0000a] in <3a61d48d66dd458fa0b64990b6c8f8cc>:0 
   at Plugin.BluetoothLE.AdvertisementData.get_ServiceUuids () [0x00000] in C:\dev\bluetoothle\Plugin.BluetoothLE\Platforms\Android\AdvertisementData.cs:29 

**Also, before catching the exception**
I found the following lines at my console:

07-21 08:39:46.693 E/ScanRecord(21533): unable to parse scan record: [11, 2, -77, -55, -9, 111]
07-21 08:39:46.703 E/ScanRecord(21533): unable to parse scan record: [11, 2, -77, -55, -9, 111]
07-21 08:39:51.764 E/ScanRecord(21533): unable to parse scan record: [-74, 20, -14, -96, -14, 90]
07-21 08:39:51.771 E/ScanRecord(21533): unable to parse scan record: [-74, 20, -14, -96, -14, 90]

bodeg, 2019-07-21:
Some more information...
Another exception while trying to

aritchie, 2019-07-21:
I’m not sure what to do when the scan result handed over is bad (as android is claiming).  Since advertisement data is lazy parsed, you can try/catch before you read them

bodeg, 2019-07-21:
Thanks for your fast reply - that's what I have done.

(should I close the issue?)

closed 2019-07-22 by aritchie