aritchie/bluetoothle #329: [Bug] Scanning doesn't work

Zarandor, 2019-07-23:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
Wiko Lenny 3 - Android 6.0

## Steps to Reproduce
``
IAdapter bleAdapter = CrossBleAdapter.Current;

//- logging before scanning -
Log("before - isScanning=" + bleAdapter.IsScanning + ", isAdvertising=" + bleAdapter.Advertiser.IsStarted + ", adapterStatus=" + AdapterStatusString(bleAdapter.Status) + ", adapterFeatures=" + AdapterFeaturesString(bleAdapter.Features));

//- scanning -
sScanner = bleAdapter.Scan(config).Subscribe(scanResult => ProcessScanResult(scanResult));

//- logging after scanning -
Log("after - isScanning=" + bleAdapter.IsScanning + ", isAdvertising=" + bleAdapter.Advertiser.IsStarted + ", adapterStatus=" + AdapterStatusString(bleAdapter.Status) + ", adapterFeatures=" + AdapterFeaturesString(bleAdapter.Features));

## Expected Behavior
Scanning should work on Android as it works on iOS.

## Actual Behavior
No devices are discovered on Android contrary to iOS (iPhone 5s - iOS 12.3.1) where it works.

Here are the log results from the code:

before - isScanning=False, isAdvertising=False, adapterStatus=PoweredOn, adapterFeatures=All

after - isScanning=True, isAdvertising=False, adapterStatus=PoweredOn, adapterFeatures=All

FYI:
- GATT server + advertising work on both device (Android & iOS).
- By using a BLE scanner app on the android, devices around were found
- Min version: Android 5.0 / Target version: Android 8.1 (= compiled version used)

Zarandor, 2019-07-23:
The Logcat (BleClient belongs to my app)

BleClient, StartScanning(), begin
BleClient, StartScanning(), ServiceId=8e6a0000-4778-a935-f823-ccf2c43314ff
BleClient, StartScanning(), deviceName=Default Bluetooth Device, MainThreadSuggested=False
D/BluetoothAdapter(31356): isEnabled
D/BluetoothAdapter(31356): 257746138: getState(). Returning 12
BleClient, StartScanning(), before - isScanning=False, isAdvertising=False, adapterStatus=PoweredOn, adapterFeatures=All
D/Mono    (31356): Loading reference 2 of Plugin.BluetoothLE.dll asmctx DEFAULT, looking for System.Reactive, Version=4.1.0.0, Culture=neutral, PublicKeyToken=94bc3704cddfc263
D/Mono    (31356): Assembly Ref addref Plugin.BluetoothLE[0xab565a40] -> System.Reactive[0xab566160]: 3
D/Mono    (31356): Loading reference 3 of Plugin.BluetoothLE.dll asmctx DEFAULT, looking for System.Core, Version=2.0.5.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e
D/Mono    (31356): Assembly Ref addref Plugin.BluetoothLE[0xab565a40] -> System.Core[0x9aea03c0]: 13
D/BluetoothManager(31356): getDevicesMatchingConnectionStates
D/BluetoothAdapter(31356): getLeState() returning 12
D/BluetoothAdapter(31356): getLeState() returning 12
D/BluetoothAdapter(31356): STATE_ON
D/BluetoothAdapter(31356): getLeState() returning 12
D/BluetoothLeScanner(31356): onClientRegistered() - status=0 clientIf=5
I/BluetoothLeScanner(31356): startRegisteration: mLeScanClients={md56e93a61007e9534b9b6779ca67ff3b68.LollipopScanCallback@ccb727e=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@d964efb}
D/BluetoothAdapter(31356): isEnabled
D/BluetoothAdapter(31356): 257746138: getState(). Returning 12
BleClient, StartScanning(), after - isScanning=True, isAdvertising=False, adapterStatus=PoweredOn, adapterFeatures=All
BleClient, StartScanning(), end
BleClient, StopScanning(), begin
D/BluetoothAdapter(31356): getLeState() returning 12
D/BluetoothAdapter(31356): getLeState() returning 12
D/BluetoothAdapter(31356): STATE_ON
I/BluetoothLeScanner(31356): startRegisteration: mLeScanClients={md56e93a61007e9534b9b6779ca67ff3b68.LollipopScanCallback@ccb727e=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@d964efb} ,callback=md56e93a61007e9534b9b6779ca67ff3b68.LollipopScanCallback@ccb727e
BleClient, StopScanning(), end

aritchie, 2019-07-24:
Please try the sample on your device.  I have no troubles using this anywhere and I don’t see anything wrong at the surface level code, but I can’t see enough to even make a guess

Zarandor, 2019-07-24:
Hi, it's the same result.
iOS: OK
Android: NOK

Do you have the apk of the android sample ? Thus, I'll test the sample built by you and not me.

Zarandor, 2019-07-24:
GPS MUST be activated!
Where is it metioned in the doc ?
So with gps activated the scan works now.

closed 2019-07-24 by Zarandor

aritchie, 2019-07-24:
It shows location permissions right under the android setup.  I even tell you what plugins to use.  If you can improve the docs, feel free to send a PR.

Zarandor, 2019-07-24:
You don't understand, setting fine and coarse permissions doesn't imply that gps have to be activated for the plugin to work. So we have to guess that which is the problem here.

aritchie, 2019-07-24:
The GPS doesn’t need to be activated unless it is something weird with that model of phone.  I have no idea.  There isn’t really a problem with the plugin and I’m glad you’ve solved your issue.

Zarandor, 2019-07-24:
One more note on this, to go with the philosophy of the plugin. You give the possibility to open the settings to activate the bluetooth, same thing then should be done for gps. I'm sure that it's not the only phone when gps is mandatory for scanning to work.

aritchie, 2019-07-24:
No thanks.  This is the first I’ve time I’ve ever encountered this issue and I don’t personally have a requirement for it.  You may be able to do it with the permission plugin

Zarandor, 2019-07-24:
Yes, the permission plugin will do it.
I saw when I launched ACR BluetootLE that permission to access phone position is asked.
I, of course replied no because why a bluetooth app will require GPS to locate other devices by Bluetooth?
Think of basic users, will they find it normal ? I make apps for the general public, not for programmers. So yes, we have to ask them for permission but we also have to explain them why and be convincing.

As a programmer, I spent a lot of time trying to figure out how to use the plugin and what was wrong so I agree with you: a serious update on the doc is required because the "How to use" and the docs are wrong or full of contradictions.

aritchie, 2019-07-24:
Ok Listen - I didn't ask you to use this plugin.  I don't owe you anything.  I offer this code up for free.  The code is all open.  I build on top of the OS platform where you can read the native docs as needed.  You are welcome to contribute docs, but if you are going to be rude - please don't post at all