aritchie/bluetoothle #331: [Bug]

shidem, 2019-08-08:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [x] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
Android 8.1
.NET Standard 2.0.3

## Steps to Reproduce
1. Create a blank Xamarin Forms application with .NET Standard selected as the code sharing strategy.
2. Select "Manage NuGet Packages" and add PluginBluetoothLE to solution.
3. Build the Android project.

## Expected Behavior
Build succeeds.

## Actual Behavior
Build fails with the following message -
"D:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1898,2): error XA2002: Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `System.Reactive`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `System.Reactive`."

## Additional Info
Package manager output -
Restoring packages for d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.iOS\TestBLE.iOS.csproj...
Restoring packages for d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.Android\TestBLE.Android.csproj...
Committing restore...
Writing lock file to disk. Path: d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.iOS\obj\project.assets.json
Restore completed in 58.51 ms for d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.iOS\TestBLE.iOS.csproj.
Successfully installed 'Plugin.BluetoothLE 6.2.6' to TestBLE.iOS
Successfully installed 'System.Reactive 4.1.5' to TestBLE.iOS
Successfully installed 'System.Runtime.InteropServices.WindowsRuntime 4.3.0' to TestBLE.iOS
Successfully installed 'System.Threading.Tasks.Extensions 4.5.2' to TestBLE.iOS
Executing nuget actions took 2.32 sec
Committing restore...
Writing lock file to disk. Path: d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.Android\obj\project.assets.json
Restore completed in 55.79 ms for d:\Users\steveh\source\repos\TestBLE\TestBLE\TestBLE.Android\TestBLE.Android.csproj.
Successfully installed 'Plugin.BluetoothLE 6.2.6' to TestBLE.Android
Successfully installed 'System.Reactive 4.1.5' to TestBLE.Android
Successfully installed 'System.Runtime.InteropServices.WindowsRuntime 4.3.0' to TestBLE.Android
Successfully installed 'System.Threading.Tasks.Extensions 4.5.2' to TestBLE.Android
Executing nuget actions took 3.86 sec
Time Elapsed: 00:00:06.5356406
========== Finished =========="

Note that this is a barebones skeleton solution to attempt to determine why the Tasks.Extensions causes build to fail. Also note that version 6.2.3 builds successfully. I have the same problem with 6.2.5 and 6.2.4.

aritchie, 2019-08-08:
Install System.Threading.Tasks.Extensions nuget into your android (and likely iOS) project if you have this problem.  It usually only happens with older versions of VS2017.

closed 2019-08-08 by aritchie

shidem, 2019-08-08:
Hi Allan,
My version of VS2017 Enterprise is 15.9.14 and I keep it maintained. I also have tried installing the System.Threading.Tasks.Extensions via NuGet, but I still get the same build error.

By the way, I really do like your library and version 6.2.3 does everything I need. I just like to keep up with bug fixes and new features/extensions.

Best regards,
Steve Hidem

aritchie, 2019-08-08:
You have to install that nuget directly into the head android & ios projects, not the netstandard.  If this still doesn't work for you, I would suggest talking to xamarin & the reactive team.  This has always solved it for me.

shidem, 2019-08-08:
Thanks, Allan! Installing System.Threading.Tasks.Extensions selecting the only the device projects did the trick. I was doing it selecting all the projects.

Thanks again,
Steve Hidem