aritchie/bluetoothle #332: [Bug] Discover Characteristics works on iOS, but not Android

tpitman, 2019-08-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
iOS 12
Android 8.0.0

## Steps to Reproduce
1. Create app that has the code shown below
2. Run on iOS the discover callback works
3. Run on Android the discover callback never gets called

## Expected Behavior
Works on iOS, not on Android

## Actual Behavior
Works on both

Here is my code:

`
                              if (!_connected)
                                    Item.Device.Connect();

                                _ = Item.Device.WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>
                                {
                                    if (characteristic.ToString().ToUpper().Equals("EB0A0002-AF09-4B25-B6AC-CD1ACA7D42E2"))
                                    {
                                        await characteristic.Write(new byte[] { 0xFF });
                                        _connected = false;
                                    }
                                });`

aritchie, 2019-08-15:
Connect is a request operation.  It could happen now or 30 mins from now. 

Please take some time to review the samples and documentation before filing something.  This is not a support site.

closed 2019-08-15 by aritchie

tpitman, 2019-08-15:
Ok. Sorry for that. I was following an example, but maybe it was wrong. I will check the samples closer.

tpitman, 2019-08-15:
I found everything I needed and now everything is working great on iOS and Android.

Sorry again for not paying better attention before submitting this issue.

You are awesome and thank you a million for the plugin.