aritchie/bluetoothle #334: [Bug] NullReferenceException on DeviceContext.Gatt when in GetKnownService

pdecostervgls, 2019-08-30:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS (Probably, no Exception being thrown)
 - [x] Android
 - [x] UWP
 - [ ] .NET Standard

## Version of Library
Plugin.BluetoothLE cebce74825490a20a39df579a084340d92422a52

## Version of OS(s) listed above with issue
Android 9

## Steps to Reproduce
```
IGattService service = await this.bluetoothDevice.GetKnownService(RedactedServiceGuid);
IGattCharacteristic redactedChar = await service.GetKnownCharacteristics(RedactedCharacteristicGuid);
await redactedChar.Write(bytes);
```
## Expected Behavior
No exceptions and an `IGattService` being returned.

## Actual Behavior
`this.context.Gatt.DiscoverServices();` throws `NullReferenceException` because `this.Context.Gatt` is `Null`

aritchie, 2019-08-30:
Please give me some sort of details if you want any sort of help!  If the gatt object is null, it probably means you aren't connected.  What actual nuget version are you using?

pdecostervgls, 2019-08-30:
@aritchie Sorry, I know the details were scarce, I sort of suspect the internal nativeDevice and gatt object get disposed because I am using this object over multiple viewmodels by means of a service.
I was using the last NuGet version available, but I actually cloned the master branch and linked that project to be able to debug this issue. Its not happening very consistenly, but it throws a wrench into my Bluetooth communication at the moment. I'd be happy to provide any details you require.

aritchie, 2019-08-31:
You have to provide some sort of reproducible sample and a stack trace please.

pdecostervgls, 2019-09-02:
My bad, I found out the issue was caused by me unintentionally disconnecting from the device. However I do believe this scenario could benefit from a more precise error message. Something along the likes of a `DeviceDisconnectedException`?

aritchie, 2019-09-02:
Glad you found your issue.  Feel free to send a PR

closed 2019-09-02 by aritchie