aritchie/bluetoothle #335: [Bug] System.Threading.Tasks.Extensions referenced by System.Reactive

sesitosvdx, 2019-09-03:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
.NETStandard 2.0.3
Android 9.0

## Steps to Reproduce
1. Create a blank Xamarin Forms application with .NET Standard selected as the code sharing strategy or create a blank app a Prism blank app (same result).
2. Install "Plugin.BluetoothLE" to all projects from the solution (v6.2.6)
3. Install "System.Threading.Tasks.Extensions" to Android project. (v4.5.2 or v4.5.3)
4. Build the Android project.

## Expected Behavior
Build succeded.

## Actual Behavior
Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `System.Reactive`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `System.Reactive`.	testnoprismBLEPlugin.Android

It is the same isssue as this: https://github.com/aritchie/bluetoothle/issues/331
I have same version of VS2017, 15.9.15, but I couldn't fix it as he did.

Thanks in advance!

aritchie, 2019-09-03:
The issue doesn't really have anything to do with this library.  If those workarounds aren't working, I suggest you talk to Xamarin or upgrade to VS2019

closed 2019-09-03 by aritchie

sesitosvdx, 2019-09-03:
Yeah, it does not really have to do with this library, but it is not letting us use it :(

If anyone else have the same problem in the future, I solved it by installing System.Threading.Tasks.Extensions v4.4.0 (only in Android - iOS project). 

Thank you anyway