aritchie/bluetoothle #336: [Bug]  Error on adding package: Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `System.Reactive`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `System.Reactive`.

KRA2008, 2019-09-11:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [x] .NET Standard

## Version of Library
6.26
## Version of OS(s) listed above with issue
n/a

## Steps to Reproduce
1. Create new app
2. Add Nuget Package
3. Try to build/deploy

## Expected Behavior
Should build/deploy

## Actual Behavior
Get error: Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `System.Reactive`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `System.Reactive`.	

ALSO, proceeding with adding the nuget package and assembly reference as suggested does not help.

aritchie, 2019-09-11:
Please do a search on closed issues.  I have answered this so many times.

closed 2019-09-11 by aritchie

KRA2008, 2019-09-11:
Ok, thank you. Switching to VS2019 fixed it for me.