aritchie/bluetoothle #337: [Bug] Gatt service Uuid

tentom, 2019-09-14:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android v9 

## Steps to Reproduce
Try to connect to a BLE device with multiple Gatt services.

## Expected Behavior
Multiple serviceUuids in the scanResult.

## Actual Behavior
The scan result (IScanResult) does returns two service Uuids where the the first has both the services is somewhat concertinaed.

[0:] 1816180d-0000-1000-8000-00805f9b34fb
[0:] 00000000-0000-0000-0000-000000000000

tentom, 2019-09-16:
I have now tested this with the same Bluetooth peripheral on iOS. 
The result is as expected, so this seems to be an Android only issue.

Glomby, 2019-11-15:
We are currently working on a BLE implementation for a BLE Device with custom GUIDs and are experiencing the same problem. We need to test against two GUIDs to identify the device. The ScanResult holds all GUIDs as expected under iOS while the it fails to retrieve more than one GUID under Android.

Changing the actual amount of advertised GUIDs in the BLE Device itself to only one works fine under Android.

taiko000, 2020-02-20:
I'm also experiencing this bug while scanning for heart rate sensors. Only Android seems to be affected. On iOS, everything works as expected.

esskar, 2020-04-03:
I am experience the same problem; two service uuids are not transmitted correctly. I can confirm that with Android 6.0 and Android 8.1