aritchie/bluetoothle #340: Unable to read data after write

suriyaKalidoss21, 2019-09-17:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [ ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
 6.2.6
## Version of OS(s) listed above with issue
iOS 11.0

## Steps to Reproduce
1. In the attached sample Write the data to the Bluetooth device
2.Enable notification called before the write 
3. Read data not updated

## Expected Behavior
   Should able to read data in read characteristic

## Actual Behavior
   data not found in read characteristic
Attached code:  

         readcharecters.EnableNotifications();

                readcharecters.WhenNotificationReceived().Subscribe(result => {

                    var test = result.Data; value not obtained

                });
            }
            if (selectedCharacteristic.CanWrite())
            {
                WriteCharacteristicAsync(selectedCharacteristic);
            }
        private async void  WriteCharacteristicAsync(IGattCharacteristic characteristic)
        {
                string writetest = "LIST_DIR,audio";
                byte[] bytes = Encoding.ASCII.GetBytes(writetest);
               await characteristic.Write(bytes);
                await Task.Delay(4000);
         }

suriyaKalidoss21, 2019-09-17:
I also tried to read after write but my characteristic '6e400003-b5a3-f393-e0a9-e50e24dcca9e' does not support reads

suriyaKalidoss21, 2019-09-17:
Please provide me the solution

suriyaKalidoss123, 2019-09-18:
Hi @aritchie ,
  Could you please provide some solution

aritchie, 2019-09-18:
This is extremely poor behavior.  Don't spam me, I don't work for you!  I've supplied a library that you can use for free.  I've already told you, I'm not tech support.  I did supply you with an answer how to solve the question you were asking, it just wasn't what you needed for the device by the sounds of it.  I also notice you are spamming other projects for answers so someone can figure this out for you:

https://github.com/xabre/xamarin-bluetooth-le/issues/384
https://github.com/xabre/xamarin-bluetooth-le/issues/379

The fact is, no one knows your BLE device.  These libraries are likely working fine with said device, but you aren't adhering to how you should interact with that device.

closed 2019-09-18 by aritchie