aritchie/bluetoothle #344: [Bug] ConnectWait() does not allow disconnection

roiko, 2019-10-03:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ ] Android
 - [ ] UWP
 - [X] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android 9

## Steps to Reproduce
1. After a device is found with IAdapter.Scan(), connect with IDevice.Connect(config);
```
ConnectionConfig c = new ConnectionConfig() { AndroidConnectionPriority = ConnectionPriority.High, AutoConnect = false };
currDev.Connect(c);
```

2. Try to disconnect with CancelConnection() and WhenDisconnected()
```
currDev.CancelConnection();
var disc = await currDev.WhenDisconnected().Take(1).ToTask();
disconnected = currDev.IsDisconnected();
```

## Expected Behavior

Disconnection is executed and notified.

## Actual Behavior
Application remains stucked on awaiting for WhenDisconnected() and connection is not closed (device led stays on)

I also tried to set AutoConnect = true, but nothing changes. The only workaround I found is in changing the method I call for the connection.

Replacing

`currDev.Connect(c);`

with

`await currDev.ConnectWait().Timeout(new TimeSpan(0,0,5)).ToTask();`

I can correctly manage the disconnection.

I wrote very very basic app [here](https://bitbucket.org/ioroiko_personal/bluetoothle/src/master/)
This app is working because I'm calling the ConnectWait() method, if you try to comment this call and uncomment the one 4 lines before, the Disconnection() won't work.
**Please note**: This app scans and enable the "CONNECT" button only if the scan result device has a name that starts with "Voi" because I didn't want to waste to much time in a complete tester. If you want to use it, you have to modify line 114

`if (scanResult.AdvertisementData.LocalName.Contains("Voi"))`

by replacing "Voi" with the name of the device you want to connect to.

aritchie, 2019-10-05:
I do this in the device tests quite a bit and it works fine.  Your events are out of order though.  You are hooking to the disconnect event AFTER calling CancelConnection.  You should hook than call.

roiko, 2019-10-07:
Hello and thanks for your reply. Honestly I started from your DeviceTests.cs file. In particular I followed the RssiTests() method where the disconnection is managed as:
```
this.device.CancelConnection();
await this.device.WhenDisconnected().Take(1).ToTask();
```

Anyway, I changed my disconnection code in this:
 
```
StaticData.CurrentDevice.WhenDisconnected().Subscribe((x) => {
                        disconnected = true;  
                    });
StaticData.CurrentDevice.CancelConnection();
while (!disconnected)
{
    await Task.Delay(200);
}
```
The hook fires, but the disconnection is not actually happening. I need to call the Connect() method since (at least I think) it is the only way to set the AutoConnect = false.

aritchie, 2019-10-07:
Ok so Connect gives you an issue, but ConnectWait is working?  Sorry, I'm not sure between the title and the actual description.

roiko, 2019-10-07:
Yes, I'm really sorry I made a mistake in the title and I fixed it now. The method I need (an it is giving me the issue) is Connect().

roiko, 2019-10-11:
In the meanwhile, do you know when a new version will be released?
I've seen commit a45e70ce294b70662cdf7f2677b9a5754d02d2f9 from pdecostervgls that could be a good workaround for my specific case.

aritchie, 2019-10-11:
Questions like this are not welcome here.  My time is my time, so the answer is "when it is done"

closed 2020-01-23 by roiko