aritchie/bluetoothle #352: Your adapter status is Unknown on iOS 13

tpitman, 2019-10-24:
- [X] iOS
 - [ ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
13

## Steps to Reproduce
1. Create new project
2. Add NSBluetoothAlwaysUsageDescription to info.plist
3. Add bluetooth-central under Required background modes array to info.plist
4. Call CrossBleAdapter.Current.Scan().Subscribe in app where you want to scan for devices
5. Note Exception in Visual Studio: System.ArgumentException: Your adapter status is Unknown

## Expected Behavior
Was working in iOS 12

## Actual Behavior
App Crashes

Here is the top line of the exception trace:

`  at Plugin.BluetoothLE.Adapter.Scan (Plugin.BluetoothLE.ScanConfig config) [0x00047] in <633c0edbef6846a08601a7fa28d2b7ba>:0 `

aritchie, 2019-10-24:
Make sure the status is ready.  There is an extension WhenReady off the adapter and ScanExtra also manages this extra step for you

closed 2019-10-24 by aritchie

tpitman, 2019-10-24:
Is this a change recently in usage? This is an app that was working until I
upgraded my iPhone to iOS 13.