aritchie/bluetoothle #353: [Bug] Adapter throws exception

Andree643, 2019-10-29:
- [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
Android 8.0 (API Level 26 - Oreo)

## Steps to Reproduce
1. Accessing `CrossBleAdapter.Current` or `CrossBleAdapter.AdapterScanner.IsSupported`

## Expected Behavior
Returns Current Adapter / bool value if the Adapter scan is supported

## Actual Behavior
Throws exception
**For CrossBleAdapter.Current**
{System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentException: [Plugin.BluetoothLE] No platform plugin found.  Did you install the nuget package in your app project as well?
  at Plugin.BluetoothLE.CrossBleAdapter.get_AdapterScanner () [0x00007] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\CrossBleAdapter.cs:14 

**CrossBleAdapter.AdapterScanner.IsSupported**
{System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentException: [Plugin.BluetoothLE] No platform plugin found.  Did you install the nuget package in your app project as well?
  at Plugin.BluetoothLE.CrossBleAdapter.get_Current () [0x00007] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE\CrossBleAdapter.cs:28

aritchie, 2019-10-29:
You need to ensure your android target meets the use nuget package target of 8.1

closed 2019-10-29 by aritchie

Andree643, 2019-10-31:
Then I guess it would be appropriate to fix the support in the readme.md where is stated version 4.3

aritchie, 2019-10-31:
4.3 is supported.  Set the target to 8.1+