aritchie/bluetoothle #354: [Bug]

lfongaroScp, 2019-10-31:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [ ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6 (vanilla, not Mvvmcross)

## Version of OS(s) listed above with issue
iOS 13

## Steps to Reproduce
1. Start a scan in a APP in debug mode
2. Close abruptly trough the IDE (Visual studio for mac)
3. Restart

## Expected Behavior
The scan start without problem

## Actual Behavior
The library crash with "there is already another scan"

I start the scanner in the MainViewController inside ViewwillAppear with ScanExtra.
My code
```
if(CrossBleAdapter.Current.IsScanning)
            {
                CrossBleAdapter.Current.StopScan();
            }

            scanner = CrossBleAdapter.Current.ScanExtra(new ScanConfig {
                ScanType = BleScanType.LowLatency,

            }).Subscribe(ManageScanResult);

```
Note the CrossBleAdapter.Current.IsScanning return always false in my test.

In ViewWillDisappear
```
if(scanner != null)
            {
                scanner.Dispose();
            }
```

ManageScanResult just add the device to a List

I already read #161 but it does not resolve my problem. I use ScanExtra as you can see. My app is in Xamarin.iOS (not Xamarin.Forms)

aritchie, 2019-10-31:
I can’t reproduce this.  IsScanning is checking directly against the cbcentralmanager so the fact that it is returning false and two lines later is returning true is odd.  

Note that you can call StopScanning without checking the flag yourself.  Scanextra also has a restart scan parameter you can use.

lfongaroScp, 2019-11-05:
Hi,

I test on another iOS device (iPhone 6 with iOS 12.4.3) it happened as well, but not always (iPhone 7 iOS 13 happened everytime I abrutly close the app from the IDE).

However, I resolved with your suggestion about the restart flag:
```
scanner = CrossBleAdapter.Current.ScanExtra(new ScanConfig {
                ScanType = BleScanType.LowLatency,
                
            }, true).Subscribe(ManageScanResult);
```
I do not know if the flag restart cause overhead, but in my case it doesn't matter. My app will scan for about 20-30 second (I need to test it against the bluetooth device I use)
I put the Dispose only in ViewWillDisapper for testing purpose.

Thanks 👍

closed 2019-11-05 by lfongaroScp