aritchie/bluetoothle #356: [Question] Check if device is turned off

bluekuen, 2019-11-14:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x ] iOS
 - [ ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.3

## Steps to Reproduce
1. Scan devices
2. update Rssi 
3. Rssi didn't change when device is turned off

## Expected Behavior
i get Rssi of 0 ?

## Actual Behavior
the last Rssi value is returned.

I update my Rssi value of my Devices after scan every 300 Milliseconds. Now when somebodyd turns the device off i can't react  because i'm not connected and have no opportunity to check if the device was turned off. Is there a way to observe if the device got powered off?

This is my rssi update observable which i need to optimize:
```
_rssiInterval = Observable.Interval(TimeSpan.FromMilliseconds(300)).Subscribe(async (long value) =>
                                {
                                    logger.Debug("Read Rssi " + Rssi + " from Device: " + zaDevice.Title);
                                    if( /* Detect if device is powered off or out of range */)
                                    {
                                        Rssi = 0;
                                        _rssiInterval.Dispose();
                                    }
                                    else
                                    {
                                    Rssi = await bleDevice.NativeDevice.ReadRssi();

                                    }
                                });
```

Kind Regards,
Bluekuen

closed 2019-11-14 by close-issue-app[bot]

aritchie, 2019-11-14:
I'm not really sure why the bot closed this.  You would have to monitor the state of the adapter.

bluekuen, 2019-11-14:
Can you open it then?
I don't know what you mean exactly. The adapter state of the Ble Device? 
How can i check it? Please note that i'm not connected, i just want to detect if the device is not in range anymore when scanning.

aritchie, 2019-11-14:
I'm not going to open this.  It isn't a bug.  
If you mean device like the bluetooth adapter on the mobile phone, you have adapter.WhenStatusChanged

If you mean the remote device, you use device.WhenStatusChange.

Please note I'm only willing to manage bugs & feature requests here.  The code is open for you to poke through . Questions like this are much better answered by going through that code.

bluekuen, 2019-11-14:
Yeah I see, then i have to open a feature request i guess. 
The Status is only Connected or Disconnected but i need something like in range or Available.
Thank you nevertheless and sorry for opening a Bug for that question.

aritchie, 2019-11-14:
What you are looking for is something like an ibeacon.  They were made for this type of thing.

If you need to do this with pure BLE, you'll need to setup a timer to say when things are "dead".  I wouldn't be adding anything like that to this library.  However, I do have a beacon library in the works over at https://github.com/shinyorg/

bluekuen, 2019-11-14:
Ah, i see. I was wondering if other people use timers, I also don't like that.
Thank you very much for the information! 
Seems like a big project you're working on, very interesting!