aritchie/bluetoothle #359: [Bug] After reconnect characteristics cannot be subscribed

lfongaroScp, 2019-12-02:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6 (latest stable)
## Version of OS(s) listed above with issue
Android Pie (9). It's Android one version, no mod.
Device: Nokia 7 plus

## Steps to Reproduce
1. Connect to a bluetooth device
2. Subscribe RX characteristic and do some write with a TX characteristic
3. Check it works
4. Power off the bluetooth device
5. Reconnect
6. Try to subscribe to the RX characteristics and do some write with a TX characteristic

## Expected Behavior
After the reconnect all should works as the first connection

## Actual Behavior
After the reconnection cannot read/write.
Extra info: after the reconnect if I try To subscribe a standard service (battery service) it throw:
"Object reference not set to an instance of an object"

I guess this is related to the #180 

I attach my service (I remove some useless code, I hope it is useful. It is an unbound Android service)

[ServiceAndroid.zip](https://github.com/aritchie/bluetoothle/files/3912301/ServiceAndroid.zip)

aritchie, 2019-12-02:
Please provide a full stack trace.  This usually only happens when you've pinned a previous characteristic from another connection.  I definitely won't be looking through this entire file.  You have to provide small reproducible cases.

lfongaroScp, 2019-12-04:
I will try to explain better the problem I face. Maybe (most likely) I do not understand something about your library.

Case:
I have a device with the standard service battery and a standard characteristic: battery level. 

In my class I have two member:
- a reference to the device
- a IDisposable reference to the subscribe of the characteristic
I get a reference to the device through a Guid I get from a previous scan

```
CrossBleAdapter.Current.GetKnownDevice(id).Subscribe((device) =>
                {
                    this.device = device;
                    ManageConnectionToDevice();
                });
```
Now, in ManageConnectionDevice I do three thing:
- subscribe to the deviceStatusChange observable.
- connect to the device

This is the code inside the method:

```

device.WhenKnownCharacteristicsDiscovered(BatteryServiceGuid,
                                BatteryStandardChargeGuid).Subscribe((characteristic) => {
                                disposableReference = characteristic.RegisterAndNotify()
                                .Subscribe(result =>
                                            {
                                                //do something with the value
                                            })
                                });

device.WhenStatusChanged().Subscribe((status) => {

                if (status == ConnectionStatus.Disconnected)
                {
                              disposableReference.Dispose();

                } 
}

device.Connect(new ConnectionConfig()
            {
                AutoConnect = true
            });
});
```

When the device is power off and power on (Disconnected and Connected) the subscribe 
```
device.WhenKnownCharacteristicsDiscovered(batteryServiceGuid,
                                batteryStandardChargeGuid).Subscribe(//code)
```
trigger, but the inside subscrition 
```
 characteristic.RegisterAndNotify().Subscribe(result =>
            {
                //do something with the value
            })
```
Do not trigger.

Same happen for a characteristic with write capability:

I hope I explain myself better.

aritchie, 2019-12-04:
I still need a full stack trace please

lfongaroScp, 2019-12-05:
HI,

I cannot submit a full stack trace because it does not throw an exception nor it trigger an error in the subscribe.
However, in the device log I found this (after the reconnection)

> 12-05 10:58:08.678: E/BtGatt.GattService(25212): registerForNotification() - No connection for <device uuid>

I attach a small complete app (Android) that reproduce the problem
[TestAppBluetooth.zip](https://github.com/aritchie/bluetoothle/files/3926194/TestAppBluetooth.zip)

Regards,

aritchie, 2019-12-06:
You said this in your first message:

Extra info: after the reconnect if I try To subscribe a standard service (battery service) it throw:
"Object reference not set to an instance of an object"

^^^ I'm looking for the full stack trace here 

Honestly, just use ConnectHook if you are looking to auto-connect & hook a battery notification.  It will manage everything for you.

nexxuno, 2019-12-07:
Unfortunately I have the same problem when using ConnectHook, I'm still trying to figure out what's going on but after the device returns online the notifications do not come back.

nexxuno, 2019-12-08:
BY looking deeper into this the connectHook is working (i.e. when the device is autoconnected the registerandnotify are fired for all the characteristics) but then again the notifications are not emitting. I'll dig deeper if I can.

lfongaroScp, 2019-12-09:
> You said this in your first message:
>
> Extra info: after the reconnect if I try To subscribe a standard service (battery service) it throw:
> "Object reference not set to an instance of an object"
>
> ^^^ [...]

Sorry, my bad. I forgot to update about it: I cannot reproduce the exception, only the problem about the missing notifications. 

Do you check my example app? It reproduce the problem we talk about and I guess it's the same nexxuno report.
My example is about the standard battery service to keep the example itself simple.
The only extra information I got is the line I found in the Android Logcat. I'm sorry, but I do not find other informations.

In the meantime I will try connectHook

Thanks for the help.

aritchie, 2019-12-10:
@IfongaroScp no - I don't look at full apps.  I'll look at small chunks of code.  You should try the sample app with this repo to see if it works for you.

lfongaroScp, 2019-12-10:
Hi
Thanks for the reply
- I submit you the small chunks of code and a minimal app. So, you have all you need, use what you prefere. A fullstacktrace is not available, because I cannot reproduce the exception, but only the error. As I said, I have no other information. 
- Your sample app in this repository seems broken or corrupt. There are a ton of missing reference/double import, the iOS project have not the forms package and so on.

Thanks for the help.

nexxuno, 2019-12-10:
Hello @aritchie,
you should be able to repro this by:
-connecting with connect hook to a device passing, let' say, 5 characteristics
-shutdown the device
-turn it back on
-notifications to do emit anything after the auto-recconnect

If you can't reproduce it this way maybe it's the android version or a problem with specific BT devices.

ThiagoAguirreLorscheiter, 2019-12-13:
Hi @aritchie!
I'm having the same problem as @nexxuno, once I shutdown the device and turn it back on notifications aren't working. Using nRF app to hook into the same connection I've noticed that the notification aren't enabled at all. If I manually turn notifications on nRF, my app start working again(works on android 9, fails on android 5), meaning that the problem isn't receiving the notification but enabling them(at least for android 9). Another point to note is if I quickly turn the device off and on again the notifications are enabled once it reconnect. Hope it helps.
Cheers!

aritchie, 2019-12-13:
It will get fixed if/when I feel like it.  I don't have the issue so everyone can quit posting unless they are going to contribute a PR.

lfongaroScp, 2020-03-30:
Hi,

After test and a lot of search a coworker of mine find a clue. It's not a library problem, but a characteristic of bluetooth in Android system, in other words: work as intended
I try to explain, but I'm not very skilled about bluetooth :-) 
Android assign, under the hood, a property to the bluetooth devices when scan. This property is not the id we get from the scan (device.Id).
This property is stored only for paired devices, otherwise is _lost_ when the user turn off the bluetooth.

So, if you use non-paired devices (like my app), you need to execute a new scan to ensure you have the correct property (the id you get from class Device is still valid). Better to set a timeout in this case.

Here a snippet:
```
                    CrossBleAdapter.Current.ScanUntilDeviceFound(idDevice).Timeout(timeout)
                    .Subscribe((device) => {
                        this.device = device;
                    }, onError: (error) => {
                        if (error is TimeoutException) //manage the type of exception
                        {

                        }
                    });
```
Obviously it's not very efficient, but I'm not very skilled.
Feel free to improve the approach, for my case this is enough.

Thanks to @aritchie for this library. 👍

closed 2020-03-30 by lfongaroScp

nexxuno, 2020-03-30:
I don't know if this bug should be closed or we should find a way to engineer this in the library itself. Maybe it doesn't make anysense? I'm thinking about connecthook for example. @aritchie any thoughts?