aritchie/bluetoothle #360: [Bug] ConnectHook

nexxuno, 2019-12-05:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android Pie

## Steps to Reproduce
Use connectHook with more than one Guid in input

`connectHook = belt.ConnectHook(ReadWriteHelper.BMAP_SVC_GUID,
                new Guid[]
                {
                    
                    ReadWriteHelper.CreateCharGuid((int)CharacteristicsIdsForRead.BT_GATT_DEBUG_3),
                    ReadWriteHelper.CreateCharGuid((int)CharacteristicsIdsForRead.BT_GATT_ERROR)
                    
                })
                //.Retry() //this doesn't help
                //.Sample(TimeSpan.FromSeconds(2)) //this doesn't help
                .Subscribe(
                x =>
                {
                    try
                    {
                        int charId = ReadWriteHelper.BackToInteger(x.Characteristic.Uuid);
                        MemoryImage.SaveValueForChar(charId, x.Data);
                    }
                    catch
                    {
//TODO
                    }
                },
                ex =>
                {
                    Crashes.TrackError(ex, new Dictionary<string, string> { { "Error from connecthook subscriptions", "Error from connecthook subscriptions" } });
                });`

## Expected Behavior
All the Characteristics receive notifications in Subscribe

## Actual Behavior
Only the last characteristic in the list receives notifications in Subscribe

Note that they look like they are ubscribed from the logs:

12-05 21:34:21.805 D/BluetoothGatt(22429): setCharacteristicNotification() - uuid: 00000013-0000-1000-8000-00805f9b34fb enable: true
...
12-05 21:34:22.249 D/BluetoothGatt(22429): setCharacteristicNotification() - uuid: 00000003-0000-1000-8000-00805f9b34fb enable: true

nexxuno, 2019-12-07:
I fixed this problem in ConnectHook but then again I have a problem with characteristics not receiving notifications after I turn off and then on the device.

Anyway I think the problem in ConnectHook is:

                var sub = device
                    .WhenConnected()
                    .Select(_ => device.WhenKnownCharacteristicsDiscovered(serviceUuid, characteristicUuids).Retry())
                    .Switch()
                    .Select(x => x.BetterRegisterAndNotify(false, true))
                    **.Merge()** //THIS HERE WAS A SWITCH SO THE LAST CHAR SUBSCRIPTION CUTS OUT THE OTHERS
                    .Subscribe(
                        ob.OnNext,
                        ob.OnError
                    );

nexxuno, 2020-01-06:
Hello, isn't my proposed fix good? Won't fix?

aritchie, 2020-01-07:
The PR is fine.  I just have other things to work on right now.  You said this didn't fix your issue entirely anyhow.

nexxuno, 2020-01-07:
Hello,
This fixes the bug decribed here. What this doesn't fix is the other problem I have descrbed in the other bug (i.E. Problems on reconnections with many characteristics subscribed).

Ottieni Outlook per Android<https://aka.ms/ghei36>

PR #362 (2020-03-30): Fix for connecthook with many characteristics
Fix for connecthook with many characteristics

### Description of Change ###
Switch took only the last characterstic in the list and cut out all other emissions, now it doesn't

### Issues Resolved ### 
- fixes #360 

### API Changes ###
 None

### Platforms Affected ### 
- All

### Behavioral Changes ###
None

### Testing Procedure ###
Use connecthook to hook to many characteristics of a BT device, let's say 5 characteristics
You'll now receive notifications from the connecthook subscription from all of them and not only from the fifth.

### PR Ch [... cut]
closed 2020-03-30 by aritchie