aritchie/bluetoothle #361: [Bug] Not able to connect without Pairing on Android

kavitap810, 2019-12-10:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [*] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6

## Version of OS(s) listed above with issue
Android 7.0 and above

## Steps to Reproduce

1. Do not implement Pair to Device method 
2. Try to connect to BLE Gatt server without pairing.
3. It doesn't get connected.
4. Move to Device settings and pair server device
5. Come back to the application and connect, it gets connected.
6. It works fine on IOS. 
Can you please let me know is there any configuration parameter needs to sett to make this work on Android as well, as my BLE server doesn't have pairing capabilities, my requirement is to connect without pair.

## Expected Behavior
1. Connect Device without pairing, it should get connected

## Actual Behavior
Connecting device, without pairing, it is not getting connected on Android

aritchie, 2019-12-10:
This isn't anything the library is causing or can change for you.  iOS usually is the one that calls for pairing when you try to work with secure characteristics.

closed 2019-12-10 by aritchie

kavitap810, 2019-12-10:
Thanks for your response.
But the same piece of code works differently on Android and IOS, and on android, it doesn't get connected until we pair it. Can you please help and provide some insight into it, that what could be the cause for such behavior.
To connect , we are using   ->   selectedDevice.Device.Connect();

aritchie, 2019-12-10:
I don't have any insight to offer.  It has nothing to do with the code in the library.  I use the API of the OS under the hood.  It is how the OS is handling it for whatever reason.