aritchie/bluetoothle #363: Android to Android IsConnectable never true

logikonline, 2019-12-26:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library 
6.2.6

## Version of OS(s) listed above with issue
8.1 and 9.0

## Steps to Reproduce
1. Using the sample, I want to detect a IsConnectable
2. Start on one device a server
3. Start on 2nd device a client

## Expected Behavior
Need the ability to do a connection only, not a pair.

## Actual Behavior
While iOS can detect the Android Server, the Android Client cannot.

Any assistance would be greatly appreciated.

aritchie, 2019-12-26:
It isn't supported right now on Android due to this plugin supporting old android 4.3+

closed 2019-12-26 by aritchie

logikonline, 2019-12-26:
If I fork it an move it to 7.1, would that resolve it?

logikonline, 2019-12-26:
> It isn't supported right now on Android due to this plugin supporting old android 4.3+

Allan, can you tell me if I fork it and rev it to 7.1 would it resolve this or is there an additional change that must occur?