aritchie/bluetoothle #364: [Bug] ble client Apk is crashing when ble connection is closed is an unexpected way from server side

falvado, 2020-01-02:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android

## Steps to Reproduce
1. Establish a BLE connection
2. Shutdown serveur when client is trying to write 
3. BLE Client is crashing

## Expected Behavior

## Actual Behavior
The application crash with this error when debugging with VisualStudio
Plugin.BluetoothLE.BleException: 'Failed to write characteristic - 133'

aritchie, 2020-01-02:
Please share some sort of code.  I can't help otherwise.  Have you tried trapping the exception?

falvado, 2020-01-03:
I am using  your plugging  to connect and exchange BLE data between a smartphone and a device (that acts as BLE server)
I can connect to the plugging and exchange data correctly  (meaning sending data and receiving notification via Rx and Tx characteristics).
The smartphone only send data to begin a process that last between 30 and 60 minutes.
During this process the smartphone receives periodically some notifications.
Most of the time everithing works ok, but if during the process I brutaly poweroff the device, BLE link is cut brutaly (and maybe not the way it should) and the error occurs.

I used :
```
        private bool SendData(byte[] data)
        {
            try
            {
                if (device_m.Status != ConnectionStatus.Connected)
                {
                    LoggerHelper.Log(LoggerHelper.eLogLevel.Info, TAG, "sendData result deconnected:" + data.AsHexString());
                    return false;
                }
                mTxCharacteristic.Write(data).Subscribe(
                            ex =>
                            {
                                if(ex.Data.AsHexString() != data.AsHexString())
                                    LoggerHelper.Log(LoggerHelper.eLogLevel.Info,TAG,   "sendData result:" + ex.Data.AsHexString() + " for sent:" + data.AsHexString());
                             });
                return true;

            }
            catch (Exception e)
            {
                LoggerHelper.Log(LoggerHelper.eLogLevel.Info,TAG,   "sendData Failed:" + data.AsHexString() + "  with reason :" + e.ToString());
                listener_m.OnBleError(e);
                return false;
            }
        }
```
But the application keep crashing in both Release and Debug mode and In debug mode I get a visual studio error popup with the message  Plugin.BluetoothLE.BleException: 'Failed to write characteristic - 133'

aritchie, 2020-01-03:
You aren't managing the error in your write.  Your try/catch won't trap the error because Subscribe is executed later.  You need to use the overloads of Subscribe to catch the error or use await on the Write observable.  Please read the docs on the reactive extensions.

closed 2020-01-03 by aritchie