aritchie/bluetoothle #365: [Bug] Setting MTU Greater than 255 fails characteristic write

tpitman, 2020-01-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
9.1

## Steps to Reproduce
1. Create embedded device with MTU > 255 supported
2. Create app that uses RequestMTU for a size greater than 255
3. Use characteristic.Write(data).Subscribe(...) to write a byte array with size > 255

## Expected Behavior
Subscribe should return in the success handler with a good result

## Actual Behavior
Subscribe returns in the error handler with the following exception text: Failed to write characteristic - 128

NOTE: on the actual device all of the data is written correctly and works fine. This is why I suspect this plugin. If you can assure me that your plugin would not cause this issue, then I will reach out to Nordic who makes the chip we are using on the device. I have control of the firmware on the device.

aritchie, 2020-01-07:
RequestMTU is a negotiation process.  Just because you request and the host supports it, it doesn't mean your device does.  

Before filing things like this, please do some debugging with the library first to verify a bug with a reproducible test case.

closed 2020-01-07 by aritchie

tpitman, 2020-01-07:
How do I do this debugging that you are talking about? I am the maker of the device and in control of the firmware. As I said in my bug report the device receives ALL of the data in the packet I am sending so as far as I can tell the device DOES support that size MTU. I would expect it wouldn't be able to get all of the bytes if that wasn't the case.

tpitman, 2020-01-07:
Another note: I am using the RequestMTU(488). It is returning 488 as the MTU. Does that mean the negotiation succeeded? If so then either there is a bug in the negotiation or there is a bug in supporting the writing of the number of bytes the negotiation reported back.

aritchie, 2020-01-07:
Device = mobile device - HOST - your firmware device.  The mobile device (especially android) often doesn't have great MTU.

> How do I do this debugging that you are talking about?

Add the source code for this library to your project or use the sample - debug in visual studio

> Another note: I am using the RequestMTU(488). It is returning 488 as the MTU. Does that mean the negotiation succeeded?

Please have a look through the source code for yourself. 
https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE/Platforms/Android/Device.cs#L204

I don't mean to be rude, but people seem to keep treating me as on-call support.  The code is all open.  Have you tried with another app like nrf?  Does it allow a greater MTU?

tpitman, 2020-01-07:
I don't think you are being rude at all. I totally understand. I am just trying to get to the bottom of this. I am not sure what your level of expertise is with BLE, so I am not sure what you can help with and what you can't.

I can't tell you how much I appreciate you doing this plugin at all. I know you don't have to.

I will see about debugging the code.

Do you have enough knowledge to have any idea how the firmware could be getting all of the bytes and yet the mobile side still reporting and error? Have you seen that before?

Thank you again for all the help.