aritchie/bluetoothle #367: [Bug] WhenNotificationReceived subscriber gives duplicate data after continuous use, and count of duplicate data increases as per request Tx operation

Pradip-Dhanraj, 2020-02-18:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ✓] iOS 
 - [ ✓] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
version 6.2.6

## Version of OS(s) listed above with issue
Android Pie (9). It's Android one version, no mod.
Device: Pixel 2 XL

## Steps to Reproduce
Connect to a Bluetooth device
Subscribe RX characteristic.
In loop, do some write with a TX characteristic to receive multiple frames of byte[20] (at least 4 frames).
WhenNotificationReceived on Rx characteristics receives same frame multiple times.
after every loop, count of duplicate frames increases.

## Expected Behavior
count of duplicate frames should not increase.

## Actual Behavior
count of duplicate frames increasing.

aritchie, 2020-02-18:
This wouldn’t happen unless you are rehooking it.  Please provide sample code

Pradip-Dhanraj, 2020-02-19:
void SingletonSubscriber()
{
bledevice.DiscoverServices().Subscribe(chars =>
            {
                if (chars.Uuid == ServiceGIUD)
                {
                    chars.GetKnownCharacteristics(ListenerGUID)).Subscribe(ackreasult =>
                    {
                        ACKCharacteristics = ackreasult;
                        ACKCharacteristics.EnableNotifications(true).Subscribe(enablenotification =>
                        {
                            ACKCharacteristics.WhenNotificationReceived().Subscribe(res =>
                            {
                               Console.WriteLine("[{0}]", string.Join(", ", res.Data));
                                //Receiving data from this Observer
                            });
                        });
                    });
                }
            });
}

ACKCharacteristics is Property.
Above Method calling once to initialize notification.

aritchie, 2020-02-19:
As a professional, could you solve the issue with this tiny bit of code?  I need a sample or you will simply have to debug this yourself because I'm not chasing after getting the answers.  

Does the sample produce this issue for you?  
Have you looked at the code to see that there really isn't anything I'm doing that could possibly cause you to get multiple messages since I just hook an event from the OS?
Have you looked at the code on your BLE device?

aritchie, 2020-03-01:
No response received.  Not enough info to work with

closed 2020-03-01 by aritchie