aritchie/bluetoothle #368: [Bug]

lordbarba, 2020-02-21:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [X ] Android
 - [ ] UWP
 - [ ] .NET Standard

Version 6.26

## Version of OS(s) listed above with issue

## Steps to Reproduce
1. Connect to a given device
2. Enable notification
3. Write to device waiting for notification

## Expected Behavior
At the moment, when device send me notification, works if notification data is smaller than mtu (158), but if it's greater o reaceive just a part of the complete packet. 
This is what i receive: 
02 30 30 30 30 30 30 30 30 30 30 30 31 30 30 30 31 30 31 34 34 b0 80 e0 d2 ce d8 d4 a0 d2 80 80 80 80 80 81 80 80 81 84 d2 80 d2 d0 80 c6 c9 cf d7 c2 d4 b0 80 b1 c3 cb d2 d3 c2 b0 80 b4 b0 b1 b2 b0 b0 b0 80 b0 b0 b0 b1 df d0 c6 80 80 c3 cb d2 d3 c2 b0 80 b4 b0 b1 b2 b0 b0 b0 80 b0 b0 b0 b1 df c3 cc 80 80 d7 c3 cf c4 c9 c3 80 c5 c1 d5 d4 cf d2 c9 80 d4 c1 c4 c5 c6 c1 d5 80 cc d4 b1 b2 b3 b4 d7 80 c4 d3 c3 d7 80 ad 90 80 80 80 89 8d 80 d7 d3 80 ce c1 cd c3 87 e4

## Actual Behavior
This is the code i use for notification:
`charRx.EnableNotifications().Subscribe(p =>
                                {
                                    // Console.WriteLine($"-----------------> charRx.EnableNotifications().Subscribe\n");
                                    

                                    charRx.WhenNotificationReceived().Subscribe(next =>
                                    {
                                        byte[] bufByte = new byte[next.Data.Length];

                                        Console.WriteLine($"LEN CHAR: {next.Characteristic.Value.Length}. DATA LEN:{next.Data.Length}");

                                        Array.Copy(next.Data, 0, bufByte, 0, bufByte.Length);
                                        Console.WriteLine($"-----------------> READ: {ByteArrayToString(bufByte)}\n");
                                        Xamarin.Forms.Device.BeginInvokeOnMainThread(() =>
                                        {
                                            CommunicatioChannelDataReceived?.Invoke(this, bufByte);
                                        });
                                    },
                                    onError =>
                                    {
                                        Console.WriteLine($"-----------------> READ ERROR: {onError.Message}\n");
                                    },
                                    () =>
                                    {
                                        Console.WriteLine($"-----------------> READ: Completed\n");
                                    });
                                });`

The complet packet: 
02 30 30 30 30 30 30 30 30 30 30 30 31 30 30 30 31 30 31 34 34 b0 80 e0 d2 ce d8 d4 a0 d2 80 80 80 80 80 81 80 80 81 84 d2 80 d2 d0 80 c6 c9 cf d7 c2 d4 b0 80 b1 c3 cb d2 d3 c2 b0 80 b4 b0 b1 b2 b0 b0 b0 80 b0 b0 b0 b1 df d0 c6 80 80 c3 cb d2 d3 c2 b0 80 b4 b0 b1 b2 b0 b0 b0 80 b0 b0 b0 b1 df c3 cc 80 80 d7 c3 cf c4 c9 c3 80 c5 c1 d5 d4 cf d2 c9 80 d4 c1 c4 c5 c6 c1 d5 80 cc d4 b1 b2 b3 b4 d7 80 c4 d3 c3 d7 80 ad 90 80 80 80 89 8d 80 d7 d3 80 ce c1 cd c3 87 e4 c3 82 cf d2 c4 c9 ce c5 c4 80 c5 c6 c1 d5 cc d4 ce 80 d5 cd c5 d2 cf b1 f1 80 b0 d2 c3 f8 80 80 80 f8 03

I tested same device in a desktop application connected with gatt and it works like a charme

Do you have any idea how to retrieve the full packet during notification?
Thanks for your kindly help

Leonardo

aritchie, 2020-02-21:
You have to chunk your packet up and send it.  I don't do this for you.

lordbarba, 2020-02-21:
Thanks for your reply.
I'm not the owner of the Gatt Server. I'm working on a Gatt client so i can recreate packet only if i receive all the chunks. If you may suggest where to put my hands in source code, i can give it a try...
Or may u suggest a more specific solution?

Thanks
Leonardo

aritchie, 2020-02-21:
There is nothing on the client side that I know of that will fix this.

closed 2020-02-21 by aritchie

lordbarba, 2020-02-25:
Update:
I've read server device and it generate a notification for each chunk. Each chunk is smaller than MTU, so form the server point of view, it is ok. Using the library i receive ONLY the first chunk.