aritchie/bluetoothle #372: [Bug]

ArifSohaib, 2020-03-25:
I am having issues reading multiple characteristics using the Android version of the library. I ran the same code in UWP and it is working fine.

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android 9, API 28

## Steps to Reproduce
1. Find the first service 
2. Write to settings characteristic in this service
3. Enable Notifications in another characteristic in same service
4. Find second service
5. Enable notifications and read characteristic from this service

## Expected Behavior
Both characteristics on which notifications are enabled should show notifications

## Actual Behavior
I can get the response after writing to the first characteristic saying the settings are properly set
but  only the characteristic from the second service shows notifications

aritchie, 2020-03-28:
Please send a reproducible sample.

ArifSohaib, 2020-03-30:
Getting the first characteristic
``` C#
//getting the first service
IGattService firstService = await scanResult.Device.GetKnownService(Guid.Parse(SensorUUIDs.UUID_POL_SERV));
                            var settingsChar = await firstService.GetKnownCharacteristics(new Guid[] { Guid.Parse(SensorUUIDs.UUID_POL_SETI) });
//sending some data which gives settings to this characteristic
await WriteSettingsChar(settingsChar);

//adding and removed this line gives the same result, I tried it in case this characteristic was notifying
//await settingsChar.DisableNotifications();

//getting the other service
var secondService = await scanResult.Device.GetKnownService(Guid.Parse(SensorUUIDs.UUID_HRA_SERV));
//getting the second characteristic
var secondChar = await firstService.GetKnownCharacteristics(new Guid[] { Guid.Parse(SensorUUIDs.UUID_HRA_DATA) });
//getting the notification characteristic from the first service
var firstChar = await secondService.GetKnownCharacteristics(new Guid[] { Guid.Parse(SensorUUIDs.UUID_POL_DATA) });
//tried both with and without MainThread from Xamarin Essentials, same result
MainThread.BeginInvokeOnMainThread(async () =>
{
   //enable the notifications of first characteristic
    await secondChar.EnableNotifications();
   //tried both with and without delay
    await Task.Delay(200);
   //enable notifications of second characteristic
    await firstChar.EnableNotifications();
 });
//FirstCharInfo is a string property that is supposed to be updated and is bound to the XAML
//this line fires indicating that the activity characteristic can notify
FirstCharInfo = $"can notify: {firstChar.CanNotify()}";
await Task.Delay(200);
//subscribe to the first notification, this one is supposed to do some processing 
//and update the FirstCharInfo property 
firstChar.WhenNotificationReceived().Subscribe(async result => { await OnFirstNoteification(result); });
await Task.Delay(200);
//subscribe to the second notification
//this one is getting some data and updating another property
secondChar.WhenNotificationReceived().Subscribe(async result => await OnSecondNoteification(result));
```

ArifSohaib, 2020-04-09:
Any update on this?

aritchie, 2020-04-09:
@ArifSohaib you tell me?  Have you done any debugging?  If you are just expecting me to fix this for you, you've come to the wrong place.  You'll have to wait until I have time/energy to look into it myself.

ArifSohaib, 2020-04-09:
I have found code examples on how to do it but they are all in Android Java https://stackoverflow.com/questions/36539473/how-to-enable-multiple-ble-characteristic-notifications-on-xamarin-android which lists two solutions that I can't figure out how to map to C# and I found out that older versions of Android only allow 1 notification but my current minimum version(API 21) allows 15 notifications based on BTA_GATTC_NOTIF_REG_MAX.