aritchie/bluetoothle #373: [Bug] callback cannot be null

lfongaroScp, 2020-03-27:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [*] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
6.2.6
## Version of OS(s) listed above with issue
Android 6.0.1

## Steps to Reproduce
Unfortunately is a bug I cannot reproduce, but I have a full stacktrace at least.

## Expected Behavior
No crash

## Actual Behavior
Library crash on StopScan

I attach the full stacktrace I receveid from my crash reporting system.
In my app the user can launch a scan and i put this three line of code before launch it:

```
if (CrossBleAdapter.Current.IsScanning) //maybe the user already launch a scan
            {
                CrossBleAdapter.Current.StopScan();
            }
```
Then I launch a scan (I use scanextra)

Sometimes, but I cannot reproduce the bug on my devices, this exception is raised:
Java.Lang.IllegalArgumentException: callback cannot be null

The full stacktrace:
[Stacktrace.txt](https://github.com/aritchie/bluetoothle/files/4395045/Stacktrace.txt)

Is it a bug in a "old" version of Android? 

P.S. I have more info about #359 can you unlock it? I understand what is the problem, it's not a library bug, but rather a not-so-well documented behavior of Android with not paired bluetooth devices.

aritchie, 2020-03-28:
Feel free to send a PR with a fix if you want.  All of my efforts have moved to the other repo as listed in the readme.

lfongaroScp, 2020-03-30:
Hi,
Unfortunately I do not have the knowledge about bluetooth to create a proper fix and, as I said, I cannot re-create the problem.
P.S.
Can you re-open #359 ? I will put the info I got and close the issue.

aritchie, 2020-03-30:
This isn't something I can look into without a proper test case.  I would suggest wrapping a try/catch around the StopScan for now.  

The other issue is unlocked