aritchie/bluetoothle #374: [Bug] When ManufacturData is set, advertisement not working on Android

esskar, 2020-04-03:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

 - [ ] iOS
 - [X] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library

6.3.0.19

## Version of OS(s) listed above with issue

Android 6.0, 8.1

## Steps to Reproduce
```
var identity = _identityService.GetId();
_bleAdapter.Advertiser.Start(new AdvertisementData
 {
    ServiceUuids = { DiscoverableServiceGuid },
    ManufacturerData = new ManufacturerData
    {
       CompanyId = myCompanyId
       Data = myCompanyData
   }
});

## Expected Behavior

Data is advertised and will be available during scan

## Actual Behavior

Service is not advertising anything

aritchie, 2020-04-03:
It's working fine for me.  There's isn't a ton of code around the advertising module.  I'd suggest debugging and checking underlying logs in the console

esskar, 2020-04-03:
ok, i found it. i get an data too long error. thanks for the patience and great library.

closed 2020-04-03 by esskar