aritchie/bluetoothle #376: Pinging unpaired BLE devices

gajo, 2020-04-05:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE! 

## Summary
Sorry if it’s wrong place, but since most current [COVID19](https://github.com/shankari/covid-19-tracing-projects) trace tracking apps suggest using BLE is the way to go whilst preserving user privacy, I was wondering if this was possible to do using the current version of this plugin (or Shiny maybe). If not could you lay out what is needed to do with the existing projects out there (Obj-C/Swift and Java/Kotlin) trying to do the same, like where to start with native library bindings (I have some experience with those having tried using some VPN and VOIP libraries), or whatever else.

## API Changes
I gather Android uses advertise data.builder and iOS CoreBluetooth APIs, which this plugin seems to reference so it probably has most APIs

## Intended Use Case
Being able to implement an app that would respect one or many proposed protocols [CEN](https://github.com/Co-Epi/CEN/#the-cen-protocol), [STRICT](https://github.com/ito-org/STRICT), [PPDT](https://github.com/degregat/ppdt)...

aritchie, 2020-04-05:
You can’t get the MAC address on most of the platforms

closed 2020-04-05 by aritchie

gajo, 2020-04-06:
Thanks forthe awsome fast reply, saw it this morning...but how then, [bluetrace.io](bluetrace.io) works, GATT, background Server/Central, or something else...sorry if I’m pestering, just would love using Xamarin(BLE/Shinny).

gajo, 2020-04-06:
Think [these guys](https://github.com/covid19risk/covidwatch-ios) confirmed the presumption but if you could confirm please and comment if that is redouble in xamarin using ble/shinny plugin I’d be grateful...if so guessing number of contacts go unnoticed unless this works really fast...but could they then range properly using Bluetooth?

aritchie, 2020-04-06:
@gajo Your question keeps changing.  You wanted the MAC address off of the device - you can't do that.  You can be unhappy with that answer if you want, but there is nothing I can do.  

I don't have time to look into other solutions.  I'm really not here for that type of thing.  If they were able to accomplish it elsewhere using BLE then this library or Shiny can also do it.