aritchie/bluetoothle #378: Ble scan result event

venkat9206, 2020-04-13:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [x] Android
 - [ ] UWP
 - [ ] .NET Standard

## Feature Library

 - [ ] Core
 - [x] BluetoothLE
 - [ ] Locations
 - [ ] HTTP Transfers
 - [ ] Notifications
 - [ ] Sensors

## Version of Library
arc.ble  version 2.0.0
## Version of OS(s) listed above with issue
8.1 Oreo

## Steps to Reproduce
1. Start/initialize new scan with CrossBleAdapter.Current.Scan(sc).Subscribe($handler$);
2. Read the advertisement data inside the handler

## Expected Behavior
scan result handler is called/invoked once after the advertisement is received

## Actual Behavior
The subscribed scan result handler is being invoked multiple times on receiving a single advertisement.

**Note:** this is more of a question on how the plugin works and how the advertisement is handled.
I am working on a project that requires 5 BLE devices to be advertising data simultaneously to the android app and I see a behavior of domination of advertisements from one or two devices alone which causes advertisements to be missed from the remaining devices. Since multiple events are raised for the same advertisement, it is causing a lot of overhead in processing the same advertisement repeatedly. Is there a solution to avoid this behavior?

Thanks

aritchie, 2020-04-13:
Please note that github issues are not appropriate for support questions.  The source code is all open for you to dig through to get your answers as needed.  I'll answer this one as a courtesy.

The handler isn't run multiple times for a single advertisement.  If the device is advertising on rapid iterations, it is quite possible you can pick it up on a single sweep by the OS.  I use RX specifically so you can filter/group/window the event stream to manage this however you want.

closed 2020-04-13 by aritchie

venkat9206, 2020-04-15:
Thank you for sharing additional info. Appreciate it. Will look through the source code. Thanks