aritchie/bluetoothle #379: Added bluetoothle to my solution so I can debug

tpitman, 2020-04-13:
## Summary
I need instructions on adding this project to my solution so I can build it and debug. I am having a problem that I can't seem to figure out and debugging this code would help me track it down.

## API Changes
No API changes needed

## Intended Use Case
I open my own solutions. I clone the bluetoothle repo into my computer (MacOS). I add something, probably the library project to my solution. I remove the Nuget package reference from my projects and add the library project as a reference. I build my solution and can debug into bluetoothle functionality.

I have tried doing this and get build errors. Those errors are about missing monoandroid and ios frameworks. I am not sure how to resolve because I am able to build my own solution just fine.

If I do a nuget restore on the dependancies then I get warnings at the bottom saying that the packages are already imported and can't be imported again.

Any ideas?

I am using Visual Studio Mac 2019 latest version.

tpitman, 2020-04-13:
FYI. I tried doing a clean checkout of your entire repo, opened the solution file and get the same errors there.

If I go into Android SDK Manager I have the Android 9.0, 8.1 and other SDKs installed already.

universaldata-eu, 2020-04-14:
i got the same errors when trying to build the bluetooth le project itself.
i analysed the error messages and in my case the reason was the usage of an old version of MSBuild.Sdk.Extras by the library (2.0.29). updating to the current version (2.0.54) solved the problem:
in the first line of Plugin.BluetoothLE/Plugin.BluetoothLE.csproj change
<Project Sdk="MSBuild.Sdk.Extras/2.0.29">
to
<Project Sdk="MSBuild.Sdk.Extras/2.0.54">
and rebuild.

closed 2020-04-14 by aritchie

tpitman, 2020-04-15:
This fixed the issue for me too. Thank you @universaldata-eu