aritchie/bluetoothle #380: [Bug] iOS Sample crashes

universaldata-eu, 2020-04-14:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ X] iOS
 - [ ] Android
 - [ ] UWP
 - [ ] .NET Standard

## Version of Library
current master branch 
## Version of OS(s) listed above with issue
iOS 13.4

## Steps to Reproduce
1. update MSBuild.Sdk.Extras of Plugin.BluetoothLE/Plugin.BluetoothLE.csproj (in line 1) from 2.0.29 to 2.0.54, otherwise the project will not build
2. start the Sample.iOS project in the sample directory of the solution

## Expected Behavior
the sample app should start and behave like the anroid sample app.

## Actual Behavior
after starting the app crashes immediately.
the complete debug output:
`2020-04-14 12:25:53.275 SamplesiOS[336:10877] Xamarin.iOS: Established connection with the IDE (fd: 5)
2020-04-14 12:25:53.276 SamplesiOS[336:10877] Xamarin.iOS: Processing: 'start profiler: no'
2020-04-14 12:25:53.276 SamplesiOS[336:10858] Xamarin.iOS: Profiler not loaded (disabled)
[HotReload] (2020-04-14 12:25:56.1): WARN: (Samples.iOS) Hot Reload disabled due to invalid linker settings
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Mono.Security.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Numerics.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Core.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Drawing.Common.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Xamarin.iOS.dll [External]
2020-04-14 12:25:53.931 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for Xamarin.iOS since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.938 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for Xamarin.Forms.Platform.iOS since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.941 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for Plugin.BluetoothLE since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.942 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for Acr.UserDialogs since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.942 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for BTProgressHUD since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.945 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for ReactiveUI since it's not needed (dynamic registration is not supported)
2020-04-14 12:25:53.945 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for Acr.XamForms since it's not needed (dynamic registration is not supported)
Thread started:  #22020-04-14 12:25:53.950 SamplesiOS[336:10858] Xamarin.iOS: Skipping assembly registration for SamplesiOS since it's not needed (dynamic registration is not supported)

Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Xml.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Xamarin.Forms.Platform.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Xamarin.Forms.Core.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Net.Http.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.ServiceModel.Internals.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Runtime.Serialization.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Xamarin.Forms.Platform.iOS.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/System.Reactive.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Plugin.BluetoothLE.dll
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/BTProgressHUD.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Acr.UserDialogs.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/DynamicData.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Splat.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/ReactiveUI.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Acr.XamForms.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Autofac.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Prism.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Xamarin.Forms.Xaml.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Prism.Forms.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Prism.Autofac.Forms.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/Samples.dll
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/SamplesiOS.exe
Loaded assembly: /private/var/containers/Bundle/Application/A253DAFC-BE02-4CB8-BAA0-F49390771C99/SamplesiOS.app/OpenTK-1.0.dll [External]
Thread started:  #3
[0:] LogHost: Initializing to normal mode
2020-04-14 12:25:54.676 SamplesiOS[336:10858] Xamarin.iOS: Unable to locate assembly 'ReactiveUI.XamForms' (culture: '')
2020-04-14 12:25:54.676 SamplesiOS[336:10858] Xamarin.iOS: Unable to locate assembly 'ReactiveUI.XamForms' (culture: '')
[... 22 lines cut ...]
	0x1841f0894 - /usr/lib/system/libsystem_platform.dylib : <redacted>
	0x1842dfe58 - /usr/lib/system/libsystem_kernel.dylib : <redacted>
	0x1842dfe88 - /usr/lib/system/
libsystem_kernel.dylib : <redacted>
	0x18a741cac - /System/Library/PrivateFrameworks/TCC.framework/TCC : <redacted>
	0x18a741bec - /System/Library/PrivateFrameworks/TCC.framework/TCC : <redacted>
	0x18a745b20 - /System/Library/PrivateFrameworks/TCC.framework/TCC : <redacted>
	0x1840babe4 - /usr/lib/system/libxpc.dylib : <redacted>
	0x1840af4b0 - /usr/lib/system/libxpc.dylib : <redacted>
	0x1841ad5a4 - /usr/lib/system/libdispatch.dylib : <redacted>
	0x18416b4f0 - /usr/lib/system/libdispatch.dylib : <redacted>
	0x1841630c8 - /usr/lib/system/libdispatch.dylib : <redacted>
	0x1841febac - /usr/lib/system/libsystem_pthread.dylib : _pthread_wqthread
	0x184201740 - /usr/lib/system/libsystem_pthread.dylib : start_wqthread

=================================================================
	Basic Fault Address Reporting
=================================================================
Memory around native instruction pointer (0x1842db7f4):0x1842db7e4  c0 03 5f d6 c0 03 5f d6 30 41 80 d2 01 10 00 d4  .._..._.0A......
0x1
842db7f4  e3 00 00 54 fd 7b bf a9 fd 03 00 91 b4 80 ff 97  ...T.{..........
0x1842db804  bf 03 00 91 fd 7b c1 a8 c0 03 5f d6 c0 03 5f d6  .....{...._..._.
0x1842db814  d0 03 80 d2 01 10 00 d4 e3 00 00 54 fd 7b bf a9  ...........T.{..
The app has been terminated.
`

aritchie, 2020-04-14:
The sample is “as is” and is not something I’ve changed to cause this problem.  It is likely something with the latest version of xamarin

closed 2020-04-14 by aritchie

universaldata-eu, 2020-04-14:
ok, but i have exactly the same crash in my app using the bluetooth le plugin. if i do not use the plugin (no functionality of the plugin is called), my app doesn't crash, so i suppose it has something to do with the plugin.
when debugging i found that the adapter status is 'unknown'. maybe there is a problem with the CBCentralManager?